openwrt/luci #7212: i18n: some translations have degenerated

jjm2473, 2024-07-26:
ref #7211

## Steps to reproduce:
1. Flash official 22.03.7 firmware
2. SSH than run:
  ```bash
  opkg update
  opkg install luci-i18n-base-zh-cn luci-i18n-opkg-zh-cn luci-i18n-firewall-zh-cn
  ```
3. Browser open luci web page

## Actual behavior:
[image]
[image]

## Expected behavior:
The text "Login", "Logout" should be translated.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.7'
DISTRIB_REVISION='r20341-591b7e93d3'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 22.03.7 r20341-591b7e93d3'
DISTRIB_TAINTS=''
```

jjm2473, 2024-07-26:
ping @systemcrash

systemcrash, 2024-07-26:
Not sure what to make of this one. Logout was changed in 23 at c86c256b8cf627dab2c2f6bd0f8f7d5971bf4cf7 and master at 02b66888ff21a1fd5d735ba038d6bb9a3c9d077a where i18n was kept. But this change was not back ported to 22 ever. Translations seemingly got clobbered somehow during an i18n update.

Anyway, this one is restored now via 18f3914e1b3148cfd82fe4f6d9489f43c9576df1

closed 2024-07-26 by systemcrash

hnyman, 2024-07-26:
> Not sure what to make of this one ... Translations seemingly got clobbered somehow during an i18n update

Well, looks like a strange/erroneous direct .po cherry-pick in fbc5fc5d0 .

That cherry-pick removed the translations of logout/login while the old LuCI strings still existed in 22.03 (and also introduced the translation for nonexisting "log out").
Then your later normal i18n sync 0219a27a0 properly regenerated the found 22.03 strings logout/login to .po files, but naturally without translations. (And it quite as naturally removed the "log out" with translation, as that source string did not exist in 22.03)

The error was simply the direct cherry-pick that caused havoc by changing translations .po fiies without any regard to the existence of the affected source strings.

There is the i18n merge master just for the backport purpose, as cherry-pick is not suitable.

systemcrash, 2024-07-27:
> > Not sure what to make of this one ... Translations seemingly got clobbered somehow during an i18n update
> 
> Well, looks like a strange/erroneous direct .po cherry-pick in [fbc5fc5](https://github.com/openwrt/luci/commit/fbc5fc5d0f6ac7df8202d28a7d1ed0b5ba8e1a38) .

Yeah weird - this is what the scripts produced.

> The error was simply the direct cherry-pick that caused havoc by changing translations .po fiies without any regard to the existence of the affected source strings.
> 
> There is the i18n merge master just for the backport purpose, as cherry-pick is not suitable.

OK - got it.