openwrt/luci #7213: Custom commands in Luci not work with Catalan language

manelestrada, 2024-07-27:
Steps to reproduce:

Install:
	luci-app-commands
	luci-i18n-base-ca
Select:
	/cgi-bin/luci/admin/system/system > Language and Style
	Language: Català (Catalan)
Try:
	/cgi-bin/luci/admin/system/commands
	Configure a command and Run
Actual behavior:
	Not result, no error.

Additional Information:
	It has been present since at least OpenWrt 23.05

dannil, 2024-07-27:
It seems that the apostrophes in translations doesn't get properly escaped in ucode templates before getting embedded in the resulting HTML page. For Catalan, the translation `Ha fallat l'ordre` is the culprit in this particular case as it terminates the ternary expression too early in [the code that constructs the output status string](https://github.com/openwrt/luci/blob/master/applications/luci-app-commands/ucode/template/commands.ut#L63-L68), which results in invalid JavaScript. Below is the rendering for Catalan:

```js
output.innerHTML = String.format(
	'<pre><strong># %h\n</strong>%h<span style="color:red">%h</span></pre>' +
	'<div class="alert-message warning">%s (Code: %d)</div>',
	st.command, st.stdout, st.stderr,
	(st.exitcode == 0) ? 'Command successful' : 'Ha fallat l'ordre',
	st.exitcode);
```

This affects every language that has apostrophes in the translations which are used in ucode templates, but maybe `luci-app-commands` hasn't been translated to other languages? Thereby falling back to English in the dev translations, which doesn't use apostrophes in this case.

systemcrash, 2024-07-27:
Made them double-quoted, as a work-around.

systemcrash, 2024-07-27:
Good find, @dannil - thanks!

jow-, 2024-07-27:
This commit does not fix the problem, it does not change the interpolated results.

closed 2024-07-27 by jow-

systemcrash, 2024-07-27:
Oh - I see what you did. Didn't know that. Thanks @jow-

commit 119fd22e (2024-07-27): luci-app-commands: replace i18n single quotes with double-quotes

Closes #7213
commit 385129bd (2024-07-27): luci-app-commands: use client side translation in JavaScript blocks

Do not interpolate translated strings on the server side, use browser
side translation calls instead to avoid potential quote escaping issues.

Fixes: #7213
Fixes: 119fd22ebf ("luci-app-commands: replace i18n single quotes with double-quotes")