openwrt/luci #7229: luci-app-usteer: SyntaxError An invalid or illegal string was specified

artemklevtsov, 2024-08-05:
## Steps to reproduce:

1. go to: Network → Interfaces → Usteer

## Actual behavior:

```
SyntaxError
An invalid or illegal string was specified
``

## Expected behavior:
Normal page load.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.4'
DISTRIB_REVISION='r24012-d8dd03c46f'
DISTRIB_TARGET='ipq806x/generic'
DISTRIB_ARCH='arm_cortex-a15_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt 23.05.4 r24012-d8dd03c46f'
DISTRIB_TAINTS=''
```
May be important: I use luci with nginx.

/etc/nginx/uci.conf:

```
# This file is re-created when Nginx starts.
# Consider using UCI or creating files in /etc/nginx/conf.d/ for configuration.
# Parsing UCI configuration is skipped if uci set nginx.global.uci_enable=false
# For details see: https://openwrt.org/docs/guide-user/services/webserver/nginx

worker_processes auto;

user root;

include module.d/*.module;

events {}

http {
        access_log off;
        log_format openwrt
                '$request_method $scheme://$host$request_uri => $status'
                ' (${body_bytes_sent}B in ${request_time}s) <- $http_referer';

        include mime.types;
        default_type application/octet-stream;
        sendfile on;

        client_max_body_size 128M;
        large_client_header_buffers 2 1k;
        server_names_hash_bucket_size 128;

        gzip on;
        gzip_vary on;
        gzip_proxied any;

        root /www;

        server { #see uci show 'nginx._lan'
                listen 443 ssl default_server;
                listen [::]:443 ssl default_server;
                server_name _lan;
                include restrict_locally;
                include conf.d/luci.locations;
                ssl_certificate /etc/nginx/conf.d/_lan.crt;
                ssl_certificate_key /etc/nginx/conf.d/_lan.key;
                ssl_session_cache shared:SSL:32k;
                ssl_session_timeout 64m;
                access_log off; # logd openwrt;
        }

        server { #see uci show 'nginx._redirect2ssl'
                listen 80;
                listen [::]:80;
                server_name _redirect2ssl;
                return 302 https://$host$request_uri;
        }

        include conf.d/*.conf;
}

# UCI_CONF_VERSION=1.2
```

systemcrash, 2024-08-05:
@Ramon00 any ideas?

systemcrash, 2024-08-05:
https://github.com/openwrt/luci/blob/ee1bf573ec94eb23f99c4c5edcd86e53e0300719/applications/luci-app-usteer/htdocs/luci-static/resources/view/usteer/usteer.js#L196

Ramon00, 2024-08-06:
What is wrong with that line?

Ramon00, 2024-08-06:
Maybe missing a dependency?

jow-, 2024-08-06:
This error happens when a string with syntax errors is passed to the `E()` function. Could be an invalid translation string or something like a weird hostname containing special chars.

systemcrash, 2024-08-06:
@Ramon00 is there a way to dump the (hostname etc) info from the command line to see what the GUI would be receiving?

Ramon00, 2024-08-06:
@artemklevtsov I escaped more strings in an update recently. This is available on master branch, could you try that one and see if it is fixed?

artemklevtsov, 2024-08-06:
How can I install it?

Ramon00, 2024-08-06:
Just download the file manually and put it in the right directory.

Ramon00, 2024-08-06:
@systemcrash ubus call luci-rpc getHostHints would generate the list. Next to that the app also tries a nslookup for the IPs which it finds

artemklevtsov, 2024-08-06:
@Ramon00 please reference the commit with fix. I don't see any changes in the `usteer.js` on master.

artemklevtsov, 2024-08-06:
> @artemklevtsov `ubus call luci-rpc getHostHints` would generate the list. Next to that the app also tries a nslookup for the IPs which it finds

```json
{"00:D8:61:D7:B6:EA":{"ipaddrs":["192.168.1.3"],"ip6addrs":["fd80:888a:fdf4::6c9","fdda:a71d:c2de::6c9","fe80::87d0:b1d9:69f0:c6c6"],"name":"*****.lan"},"22:4E:7F:5B:28:23":{"ipaddrs":["192.168.1.250"],"ip6addrs":["fe80::204e:7fff:fe5b:2823"]}}
```

Ramon00, 2024-08-06:
[https://github.com/openwrt/luci/blob/master/applications/luci-app-usteer/htdocs/luci-static/resources/view/usteer/usteer.js](url)
vs.
[https://github.com/openwrt/luci/blob/openwrt-23.05/applications/luci-app-usteer/htdocs/luci-static/resources/view/usteer/usteer.js](url)

Ramon00, 2024-08-06:
Also that hosthints output looks fine,assuming the **** does not contain any weird characters?

artemklevtsov, 2024-08-06:
> Also that hosthints output looks fine,assuming the **** does not contain any weird characters?
This is valid hostname (ascii, digits, dash).

artemklevtsov, 2024-08-06:
I tried this:

```bash
cd /tmp/
wget https://raw.githubusercontent.com/openwrt/luci/master/applications/luci-app-usteer/htdocs/
luci-static/resources/view/usteer/usteer.js
cp /www/luci-static/resources/view/usteer/usteer.js usteer.js.bak
cp usteer.js /www/luci-static/resources/view/usteer/usteer.js
```
Then reload page with clear cache. Nothing has changed. May be I doing something wrong.

Ramon00, 2024-08-06:
you  may need to clear  your browser cache.

Ramon00, 2024-08-06:
Could be it was not the issue. Are you using english?

artemklevtsov, 2024-08-06:
Yes, luci in english. I have 3 routers (various models and arch) and problem happens for everyone.

Ramon00, 2024-08-06:
Are you running nginx on all? Try without and see if that works.  Otherwise the only thing I can suggest is start removing pieces of code to narrow the issue down, e.g. comment out e.g. `collectRemoteHosts(remotehosttableentries,Remotehosts); `
[https://github.com/openwrt/luci/blob/master/applications/luci-app-usteer/htdocs/luci-static/resources/view/usteer/usteer.js#L225](url) and 
https://github.com/openwrt/luci/blob/master/applications/luci-app-usteer/htdocs/luci-static/resources/view/usteer/usteer.js#L344[]()

Similarly you can comment out `collectWlanAPInfoEntries` , `collectWlanAPInfos` and `collectHearingClient`

artemklevtsov, 2024-08-10:
The same error with uhttpd.

artemklevtsov, 2024-08-10:
Does anyone else reproduce this error? Maybe I have something configured wrong?

artemklevtsov, 2024-08-10:
After commenting out all calls of the mentioned functions error illegal error still present.

systemcrash, 2024-08-10:
Even an incorrect config should not cause this. Just gotta figure out
what's happening.

artemklevtsov, 2024-08-10:
I return the original file. Trace with Chromium differ than Firefox:

[192.168.1.1-1723304586716.log](https://github.com/user-attachments/files/16570118/192.168.1.1-1723304586716.log)
[console-export-2024-8-10_22-45-10.txt](https://github.com/user-attachments/files/16570119/console-export-2024-8-10_22-45-10.txt)

Ramon00, 2024-08-11:
Line 59, 60 is this:		
```
var body = E([
			E('h3', _('Hearing map')),
			E('div', _('Refresh page to get new mac addresses to show up'))
		]);
```
Try: 
```
var body = E([
			E('h3', 'Hearing map'),
			E('div', 'Refresh page to get new mac addresses to show up')
		]);
```
If that now generates an error at a different line number then you have an issue with your translation file/setup.

systemcrash, 2024-09-14:
Any follow-up here?

Ramon00, 2024-09-14:
i cant reproduce, and hence needs trouble shooting by the submitter, or somebody else that can reproduce. tbh it looks like a generic translation file issue, maybe wrong version or something

artemklevtsov, 2024-09-14:
I have this problem on 3 routers with the english luci only installation. All routers have OpenWRT 23.05.4.
usteer	2022-08-18-7d2b17c9-1
luci-app-usteer	git-24.233.60980-ae3bc90
I tried Firefox and Chromium with the same results.
I also tried files from this thread without success.

Ramon00, 2024-09-14:
it is still going through the translation system I think even if it is english. What happens if you change the line59 60 above? does the line number of the error change?

artemklevtsov, 2024-09-14:
I found reason! It's theme. Error occurs with OpenWrt theme. OpenWrt2020 is ok.

Ramon00, 2024-09-14:
nice, guess its a matter of finding the difference... Need a theme expert!

Ramon00, 2024-09-15:
little bit debugging on my part (now that i can reproduce the bug). 
I get the following error with that theme: 
```
SyntaxError
Failed to set the 'innerHTML' property on 'Element': The provided markup is invalid XML, and therefore cannot be inserted into an XML document.
```
So the default theme uses html and that Openwrt theme uses XHTML+XM, looks like it is related to that. 
https://stackoverflow.com/questions/27824609/uncaught-syntaxerror-failed-to-set-the-innerhtml-property-on-element-the-p

No sure what part exactly trying to set something invalid though.

Ramon00, 2024-09-15:
@systemcrash Maybe time to retire that old theme, or update it HTML?

Ramon00, 2024-09-15:
ok i have a possible fix in `header.ut`
replace
`
	http.prepare_content("application/xhtml+xml");
-%}

<?xml version="1.0" encoding="utf-8"?>
<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Strict//EN" "http://www.w3.org/TR/xhtml1/DTD/xhtml1-strict.dtd">
<html xmlns="http://www.w3.org/1999/xhtml" xml:lang="{{ dispatcher.lang }}" lang="{{ dispatcher.lang }}">`

with

`
	http.prepare_content('text/html; charset=UTF-8');
-%}

<!DOCTYPE html>
<html lang="{{ dispatcher.lang }}" {{ darkpref ? `data-darkmode="${darkpref}"` : '' }}>`

i can make a PR, but not sure if there will be any other fallout of this... Anybody can comment?

systemcrash, 2024-09-15:
Themes are a necessary thorn in the side and I hate xml with a passion.
Kill it with fire.

Ramon00, 2024-09-15:
https://github.com/openwrt/luci/pull/7283

have a look please

systemcrash, 2024-09-15:
Possibly fixed by #7238

closed 2024-09-15 by systemcrash