openwrt/luci #7231: luci-app-mwan3: luci shouldn't allow configuration of rule names beyond 15 characters

vincejv, 2024-08-07:
## Steps to reproduce:

1. Navigate to Network > MultiWAN Manager
2. Navigate to Rule Tab
3. Add a new rule, with a rule name greater than 15 characters example in screenshot is '1234567890123456890' (20 chars)

## Actual behavior:
Luci happily accepts the input and creates a new MWAN3 Rule
[image]
[image]

## Expected behavior:
Add a character limit on the text box input OR should prompt an error `Rule XXXXX exceeds max of 15 chars.` based on mwan3 logs

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.4'
DISTRIB_REVISION='r24012-d8dd03c46f'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 23.05.4 r24012-d8dd03c46f'
DISTRIB_TAINTS=''
```

mwan versions
luci-app-mwan3	git-22.181.29827-675a0ea	
mwan3	2.11.8-1

mwan3 does not apply rules that exceed 15 characters, sample log lines here:
```
Wed Aug  7 18:16:33 2024 user.warn mwan3-init[1844833]: Rule fastlane_rule_v4 exceeds max of 15 chars. Not setting rule
Wed Aug  7 18:16:33 2024 user.warn mwan3-init[1844833]: Rule fastlane_rule_v6 exceeds max of 15 chars. Not setting rule
```

systemcrash, 2024-08-07:
Seems like a better solution would be to increase the limit in mwan3. But I don't know enough about it to comment on that.

vincejv, 2024-08-08:
15 char limit is generous as ipset chain names are limited to 29 characters only and mwan3 appends a prefix **"mwan3_policy_"** when adding the rule in iptables, each rule has its own chain
[image]

yes mwan3 still uses iptables.. it cant simply be increased as it's a limitation on iptables, unless someone migrates the thing to nftables, which has been in progress for 2 years and counting: https://forum.openwrt.org/t/mwan3-nftables-porting-help/120895

vincejv, 2024-08-08:
also this seems like a regression?? this was discussed and fixed before: 

- https://github.com/openwrt/luci/commit/678c0b2d5ba77092ca4c422b6a3c3e3eefe86b01
- https://github.com/openwrt/luci/commit/8c19a2e1b8f8bb5f28beadd71bab88d8f892d28d
- https://github.com/openwrt/packages/issues/13499

dannil, 2024-08-08:
Seems it was left out when migrating the app to JS in [f5f6b3e](https://github.com/openwrt/luci/commit/f5f6b3e4f8e3faed4e9c9926e283c9cf226d4bd8#diff-8f78b991c5a187c3ec8c5adb877ac4849afbb76824acc0635d1c0ea3e804bd2fR29-R32), it needs to have a `validate` function which checks the length.

vincejv, 2024-08-11:
based on the two commits referenced and Github issue https://github.com/openwrt/packages/issues/13499, yes this character limit only applies to
Policies - https://github.com/openwrt/luci/commit/8c19a2e1b8f8bb5f28beadd71bab88d8f892d28d
Rules - https://github.com/openwrt/luci/commit/678c0b2d5ba77092ca4c422b6a3c3e3eefe86b01

commit 94f187bc (2024-09-14): luci-app-mwan3: add name length error check and name collision check

Closes #7231
closed 2024-09-14 by systemcrash