openwrt/luci #7235: luci-app-statistics: Logout when trying to display graphs

marek22k, 2024-08-11:
## Steps to reproduce:
**Example: Replace the following lines and remove this**

1. go to: Network → Statistics → Graphs

Since I find the behavior difficult to describe, I recorded a video of it:
[luci.webm](https://github.com/user-attachments/assets/9c01c265-03fb-469d-b37a-e0765a89127d)

## Actual behavior:

Instead of displaying the graphs, I am logged out. After a few attempts, it is no longer possible to log in.

## Expected behavior:

The graphs are displayed. I am not logged out. I can log in again.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.4'
DISTRIB_REVISION='r24012-d8dd03c46f'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 23.05.4 r24012-d8dd03c46f'
DISTRIB_TAINTS=''
```

```
root@homeguard:~# opkg info luci-app-statistics
Package: luci-app-statistics
Version: git-24.034.36441-feee897
Depends: libc, luci-base, collectd, rrdtool1, collectd-mod-rrdtool, collectd-mod-iwinfo, collectd-mod-cpu, collectd-mod-memory, collectd-mod-interface, collectd-mod-load, collectd-mod-network
Status: install user installed
Architecture: all
Conffiles:
 /etc/config/luci_statistics ad454e3b3c2c01bb29c17e275aa53e81f3f16d018a950930f138c10000d1eabf
Installed-Time: 1722985100
```
```
root@homeguard:~# cat /etc/config/luci_statistics 

config statistics 'collectd'
	option BaseDir '/var/run/collectd'
	option PIDFile '/var/run/collectd.pid'
	option PluginDir '/usr/lib/collectd'
	option TypesDB '/usr/share/collectd/types.db'
	option Interval '30'
	option ReadThreads '2'
	option Include '/etc/collectd/conf.d/*.conf'

config statistics 'rrdtool'
	option default_timespan '2hour'
	option image_width '600'
	option image_height '150'
	option image_path '/tmp/rrdimg'

config statistics 'collectd_rrdtool'
	option enable '1'
	option DataDir '/tmp/rrd'
	option RRARows '288'
	option RRASingle '1'
	option backup '1'
	list RRATimespans '2hour'
	list RRATimespans '1day'
	list RRATimespans '1week'
	list RRATimespans '1month'
	list RRATimespans '1year'

config statistics 'collectd_csv'
	option enable '0'
	option StoreRates '0'
	option DataDir '/tmp'

config statistics 'collectd_email'
	option enable '0'
	option SocketFile '/var/run/collectd/email.sock'
	option SocketGroup 'nogroup'

config statistics 'collectd_logfile'
	option enable '0'
	option LogLevel 'notice'
	option File '/var/log/collectd.log'
	option Timestamp '1'

config statistics 'collectd_network'

config statistics 'collectd_syslog'
	option enable '0'
	option LogLevel 'warning'
	option NotifyLevel 'WARNING'

config statistics 'collectd_unixsock'
	option enable '0'
	option SocketFile '/var/run/collectd/query.sock'
	option SocketGroup 'nogroup'

config statistics 'collectd_apcups'
	option enable '0'
	option Host 'localhost'
	option Port '3551'

config statistics 'collectd_chrony'
	option Host '127.0.0.1'
	option Port '323'
	option Timeout '2'
	option enable '1'

config statistics 'collectd_conntrack'
	option enable '1'

config statistics 'collectd_contextswitch'
	option enable '0'

config statistics 'collectd_cpu'
	option enable '1'
	option ReportByCpu '1'
	option ReportByState '1'
	option ShowIdle '0'
	option ValuesPercentage '1'

config statistics 'collectd_cpufreq'
	option enable '0'

config statistics 'collectd_curl'
	option enable '0'

config statistics 'collectd_df'
	option enable '0'
	option Devices '/dev/mtdblock/4'
	option MountPoints '/overlay'
	option FSTypes 'tmpfs'
	option IgnoreSelected '0'
	option ValuesPercentage '0'

config statistics 'collectd_dhcpleases'
	option enable '0'
	option Path '/tmp/dhcp.leases'

config statistics 'collectd_disk'
	option enable '0'
	option Disks 'hda1 hdb'
	option IgnoreSelected '0'

config statistics 'collectd_dns'
	option enable '1'
	list IgnoreSources '127.0.0.1'
	list IgnoreSources '::1'
	list Interfaces 'br-dmz'
	list Interfaces 'br-lan'
	list Interfaces 'laplace'
	list Interfaces 'pppoe-upstream'

config statistics 'collectd_entropy'
	option enable '1'

config statistics 'collectd_exec'

config statistics 'collectd_interface'
	option enable '1'
	list Interfaces 'br-dmz'
	list Interfaces 'br-lan'
	list Interfaces 'pppoe-upstream'

config statistics 'collectd_ipstatistics'
	option enable '1'

config statistics 'collectd_iptables'
	option enable '0'

config collectd_iptables_match
	option table 'nat'
	option chain 'luci_fw_postrouting'
	option target 'MASQUERADE'
	option source '192.168.1.0/24'
	option outputif 'br-ff'
	option name 'LAN-Clients traffic'

config collectd_iptables_match
	option chain 'luci_fw_postrouting'
	option table 'nat'
	option target 'MASQUERADE'
	option source '10.61.230.0/24'
	option outputif 'br-ff'
	option name 'WLAN-Clients traffic'

config statistics 'collectd_irq'
	option enable '0'
	option Irqs '2 3 4 7'

config statistics 'collectd_iwinfo'
	option enable '1'

config statistics 'collectd_load'
	option enable '1'

config statistics 'collectd_memory'
	option enable '1'
	option HideFree '0'
	option ValuesAbsolute '1'
	option ValuesPercentage '0'

config statistics 'collectd_netlink'
	option enable '0'
	option IgnoreSelected '0'
	option VerboseInterfaces 'br-lan'
	option QDiscs 'br-lan'

config statistics 'collectd_nut'
	option enable '0'
	list UPS 'myupsname'

config statistics 'collectd_olsrd'
	option enable '0'
	option Port '2006'
	option Host '127.0.0.1'

config statistics 'collectd_openvpn'
	option enable '0'

config statistics 'collectd_ping'
	option TTL '127'
	option Interval '30'
	option enable '1'
	list Hosts '::1'
	list Hosts 'ping.ripe.net'
	list Hosts 'fd42:d42:d42:54::1'
	list Hosts 'fd42:d42:d42:53::1'
	list Hosts 'fd42:4242:2601:ac05::1'
	list Hosts 'fd04:234e:fc31::'
	list Hosts '2a02:3001::209'
	list Hosts '62.52.192.120'
	option AddressFamily 'any'

config statistics 'collectd_processes'
	option enable '0'
	option Processes 'uhttpd dnsmasq dropbear'

config statistics 'collectd_sensors'
	option enable '1'

config statistics 'collectd_snmp6'
	option enable '0'
	option Interfaces 'br-lan'
	option IgnoreSelected '0'

config statistics 'collectd_splash_leases'
	option enable '0'

config statistics 'collectd_tcpconns'
	option ListeningPorts '0'
	option AllPortsSummary '0'
	option enable '1'
	list LocalPorts '22'
	list RemotePorts '22'

config statistics 'collectd_thermal'
	option enable '1'

config statistics 'collectd_uptime'
	option enable '0'
```

marek22k, 2024-08-11:
```
root@homeguard:~# logread | grep collectd
Sun Aug 11 07:26:31 2024 daemon.err collectd[7971]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/homeguard/chrony/time_offset-194.94.95.123.rrd: opening '/tmp/rrd/homeguard/chrony/time_offset-194.94.95.123.rrd': No such file or directory
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: Filter subsystem: Built-in target `write': Dispatching value to all write plugins failed with status -1.
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: Filter subsystem: Built-in target `write': Some write plugin is back to normal operation. `write' succeeded.
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/homeguard/chrony/clock_stratum-[UNKNOWN].rrd: illegal attempt to update using time 1723361221 when last update time is 1723361221 (minimum one second step)
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/homeguard/chrony/clock_state-[UNKNOWN].rrd: illegal attempt to update using time 1723361221 when last update time is 1723361221 (minimum one second step)
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/homeguard/chrony/clock_mode-[UNKNOWN].rrd: illegal attempt to update using time 1723361221 when last update time is 1723361221 (minimum one second step)
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/homeguard/chrony/clock_reachability-[UNKNOWN].rrd: illegal attempt to update using time 1723361221 when last update time is 1723361221 (minimum one second step)
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/homeguard/chrony/clock_last_meas-[UNKNOWN].rrd: illegal attempt to update using time 1723361221 when last update time is 1723361221 (minimum one second step)
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/homeguard/chrony/clock_skew_ppm-[UNKNOWN].rrd: illegal attempt to update using time 1723361221 when last update time is 1723361221 (minimum one second step)
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: Filter subsystem: Built-in target `write': Dispatching value to all write plugins failed with status -1.
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: uc_update: Value too old: name = homeguard/chrony/clock_last_meas-[UNKNOWN]; value time = 1723361221.481; last cache update = 1723361221.482;
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: Filter subsystem: Built-in target `write': Dispatching value to all write plugins failed with status -1.
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: Filter subsystem: Built-in target `write': Some write plugin is back to normal operation. `write' succeeded.
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: Filter subsystem: Built-in target `write': Dispatching value to all write plugins failed with status -1.
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: Filter subsystem: Built-in target `write': Some write plugin is back to normal operation. `write' succeeded.
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: Filter subsystem: Built-in target `write': Some write plugin is back to normal operation. `write' succeeded.
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/homeguard/chrony/frequency_error-[UNKNOWN].rrd: illegal attempt to update using time 1723361221 when last update time is 1723361221 (minimum one second step)
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: Filter subsystem: Built-in target `write': Dispatching value to all write plugins failed with status -1.
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: Filter subsystem: Built-in target `write': Some write plugin is back to normal operation. `write' succeeded.
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: Filter subsystem: Built-in target `write': Dispatching value to all write plugins failed with status -1.
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: Filter subsystem: Built-in target `write': Some write plugin is back to normal operation. `write' succeeded.
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/homeguard/chrony/clock_stratum-[UNKNOWN].rrd: illegal attempt to update using time 1723361221 when last update time is 1723361221 (minimum one second step)
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/homeguard/chrony/clock_state-[UNKNOWN].rrd: opening '1723361221:nan': No such file or directory
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/homeguard/chrony/clock_last_meas-[UNKNOWN].rrd: illegal attempt to update using time 1723361221 when last update time is 1723361221 (minimum one second step)
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/homeguard/chrony/time_offset-[UNKNOWN].rrd: illegal attempt to update using time 1723361221 when last update time is 1723361221 (minimum one second step)
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/homeguard/chrony/clock_skew_ppm-[UNKNOWN].rrd: illegal attempt to update using time 1723361221 when last update time is 1723361221 (minimum one second step)
[... cut ...]
	list Hosts '62.52.192.120'
	option AddressFamily 'any'

config statistics 'collectd_processes'
	option enable '0'
	option Processes 'uhttpd dnsmasq dropbear'

config statistics 'collectd_sensors'
	option enable '1'

config statistics 'collectd_snmp6'
	option enable '0'
	option Interfaces 'br-lan'
	option IgnoreSelected '0'

config statistics 'collectd_splash_leases'
	option enable '0'

config statistics 'collectd_thermal'
	option enable '1'

config statistics 'collectd_uptime'
	option enable '0'
```

marek22k, 2024-08-11:
Two other things:
1) After I set `option backup '0'`, it works again.
2) In the meantime there was another problem with collectd. I had a configuration for `collectd-mod-tcpconns` in the UCI, although I had uninstalled the plugin. This caused the same problems (if I observed it correctly).

hnyman, 2024-08-11:
How about disabling the chrony plugin that seems to flood your log with errors...
And only then trying to see what errors are left.

(e.g. trying to do backup of somehow illegally named files might fail.)

marek22k, 2024-08-11:
The whole thing seems to work without the chrony plugin. Then there are still two errors:
1) The chrony plugin does not work completely.
2) LuCi should not log me out if collectd does not work. (This also happens, for example, if a plugin is configured as described above, which is not installed).

hnyman, 2024-08-11:
> The chrony plugin does not work completely.

Well, then I would assume that your chrony config has some errors causing collectd to choke:
I would start getting rid of the UNKNOWN
```
Sun Aug 11 07:26:31 2024 daemon.err collectd[7971]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/homeguard/chrony/time_offset-194.94.95.123.rrd: opening '/tmp/rrd/homeguard/chrony/time_offset-194.94.95.123.rrd': No such file or directory
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: Filter subsystem: Built-in target `write': Dispatching value to all write plugins failed with status -1.
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: Filter subsystem: Built-in target `write': Some write plugin is back to normal operation. `write' succeeded.
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/homeguard/chrony/clock_stratum-[UNKNOWN].rrd: illegal attempt to update using time 1723361221 when last update time is 1723361221 (minimum one second step)
Sun Aug 11 07:27:01 2024 daemon.err collectd[7971]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/homeguard/chrony/clock_state-[UNKNOWN].rrd: illegal attempt to update using time 1723361221 when last update time is 1723361221 (minimum one second step)
```

The `illegal attempt to update using time 1723361221 when last update time is 1723361221 (minimum one second step)` makes me to think that there is something wrong with your clock and/or the NTP settings (apparently chrony in useas NTP). Might just be something in your chrony config causing collectd to mis-parse the data.

Collectd and thus also LuCI statistics have only limited recovery possibilities from bad data, as they are rather streamlined, trusting the user to notice config problems and then fix them.

marek22k, 2024-08-11:
The following are my chrony settings:
```
root@homeguard:~# cat /etc/config/chrony
config makestep 'makestep'
	option threshold '1'
	option limit '3'

config pool
    option hostname 'ptbtime1.ptb.de'
    option iburst 'yes'
    option nts 'yes'
    option prefer 'yes'

config pool
    option hostname 'ptbtime2.ptb.de'
    option iburst 'yes'
    option nts 'yes'
    option prefer 'yes'

config pool
    option hostname 'ptbtime3.ptb.de'
    option iburst 'yes'
    option nts 'yes'
    option prefer 'yes'

config pool
    option hostname 'ptbtime4.ptb.de'
    option iburst 'yes'
    option nts 'yes'
    option prefer 'yes'

config pool
    option hostname 'ntppool1.time.nl'
    option iburst 'yes'
    option nts 'yes'

config pool
    option hostname 'ntppool2.time.nl'
    option iburst 'yes'
    option nts 'yes'

config pool
    option hostname 'time.cloudflare.com'
    option iburst 'yes'
    option nts 'yes'

config pool
    option hostname 'pool.ntp.org'
    option iburst 'yes'

config dhcp_ntp_server
    option iburst 'yes'
    option disabled 'no'

config allow
    option interface 'lan'
    option interface 'dmz'

config makestep
    option threshold '1.0'
    option limit '3'

config nts
    option rtccheck 'yes'
    option systemcerts 'yes'
```