openwrt/luci #7245: luci-app-statistics: bug for collectd_iptable in /usr/libexec/stat-genconfig

lupien, 2024-08-17:
## Steps to reproduce:
1. have installed at least luci-app-statistics, collectd-mod-iptables, but other collectd-mod-* are useful
2. Configure Statistics → Setup with Firewall in Network activated
           To actually configure entries, you need to use iptables-legacy (from xtables-legacy). I put rules in /etc/firewall.user
3. Go to Statistics → Graph

## Actual behavior:
1. Nothing is shown. There might be an error message (depending if you have previous data or not).
The logs show errors like
```
Sat Aug 17 19:01:32 2024 daemon.err collectd[11383]: plugin_load: plugin "cpu" successfully loaded.
Sat Aug 17 19:01:32 2024 daemon.err collectd[11383]: plugin_load: plugin "interface" successfully loaded.
Sat Aug 17 19:01:32 2024 daemon.err collectd[11383]: Initialization complete, entering read-loop.
Sat Aug 17 19:01:33 2024 daemon.err collectd[11383]: plugin_dispatch_values: No write callback has been registered. Please load at least one output plugin, if you want the collected data to be stored.
Sat Aug 17 19:01:33 2024 daemon.err collectd[11383]: Filter subsystem: Built-in target `write': Dispatching value to all write plugins failed with status 2 (ENOENT). Most likely this means you didn't load any write plugins.
Sat Aug 17 19:01:33 2024 daemon.err collectd[11383]: Available write targets:: [none]
Sat Aug 17 19:01:33 2024 daemon.err collectd[11383]: Available write targets:: [none]

```

And executing "service luci_statistics restart" from the command line shows the following:
```
Reference error: access to undeclared variable rule
In config_iptables(), file /usr/libexec/stat-genconfig, line 199, byte 65:
  called from function section (/usr/libexec/stat-genconfig:251:35)
  called from anonymous function (/usr/libexec/stat-genconfig:284:17)

 `                if (match(tname, /^\S+$/) && match(chain, /^\S+$/) && match(rule, /^\S+$/) && match(name, /^\S+$/)) {`
  Near here ------------------------------------------------------------------^

```

## Expected behavior:
The graphs should work and display something, including a tab for Firewall.

## Additional Information:
My configuration was working in 22.03 but not in 23.05.
Disabling the Firewall statistics lets the other statistics work correctly.

I diagnosed the problem as a bug in the /usr/libexec/stat-genconfig script.
Here is the diff that fixes the problem

```
--- stat-genconfig.bak  2024-08-11 14:49:58.000000000 -0400
+++ stat-genconfig      2024-08-17 18:25:53.000000000 -0400
@@ -196,7 +196,7 @@
                                let tname = `${s.table}`;
                                let chain = `${s.chain}`;

-                               if (match(tname, /^\S+$/) && match(chain, /^\S+$/) && match(rule, /^\S+$/) && match(name, /^\S+$/)) {
+                               if (match(tname, /^\S+$/) && match(chain, /^\S+$/)) {
                                        str += `\t${verb} "${tname}" "${chain}"`;

                                        let rule = `${s.rule}`;

```

systemcrash, 2024-09-14:
Yeah, this looks like a variable scope issue, almost certainly. 

Please test the following code block as a replacement for the existing one and report whether it fixes your problem.

```js

function config_iptables(c) {
	let str = "";

	for (let k, s in sections) {
		for (let type, verb in { collectd_iptables_match: 'Chain', collectd_iptables_match6: 'Chain6' }) {
			if (s['.type'] == type) {
				let tname = `${s.table}`;
				let chain = `${s.chain}`;

				if (match(tname, /^\S+$/) && match(chain, /^\S+$/) {
					str += `\t${verb} "${tname}" "${chain}"`;
					let rule = `${s.rule}`;

					if (match(rule, /^\S+$/)) {
						str += ` "${rule}"`;

						let name = `${s.name}`;

						if (match(name, /^\S+$/))
							str += ` "${name}"`;
						}

						str += '\n';
				}
			}
		}
	}

	return str;
}
```

closed 2024-09-14 by systemcrash

hnyman, 2024-09-15:
I added  a fix in b9751888a

systemcrash, 2024-09-15:
Great. Thank you.

I added a fix in b975188
> <https://github.com/openwrt/luci/commit/b9751888a1f08045e589a5e3c83094a81319bf0e>
>
>
>

commit b4a25a19 (2024-09-14): luci-app-statistics: Fix variable scope

Closes #7245