openwrt/luci #7252: luci-mod-network: typo in wireless.js

efahl, 2024-08-24:
This is a trivial one for anyone who is already working on wireless.js:

The non-breaking space character in the badge output has its digits transposed, it's `\x0a` and it should be `\xa0`.  This causes the badge to be rendered sometimes as two lines, sometimes as three and the jumpy output is pretty annoying.  (If you scan the file for `\xa0`, you'll see the other strings that are properly formatted.)

Here's the line in question:
```
			value = '---/%d\x0a%s'.format(noiseValue, _('dBm'));
```

Currently at

https://github.com/openwrt/luci/blob/master/modules/luci-mod-network/htdocs/luci-static/resources/view/network/wireless.js#L83

Ramon00, 2024-09-20:
is this still an issue? Want me to make a PR? That is like 1 min work...

systemcrash, 2024-09-20:
Yes please 👌

systemcrash, 2024-09-20:
Fixed. :)

closed 2024-09-20 by systemcrash

Ramon00, 2024-09-20:
ah very good!

commit a2c21a62 (2024-09-20): luci-mod-network: change to non-breaking space character

Closes #7252

Thanks to @efahl
commit 63ba3cba (2024-09-20): luci-mod-network: change to non-breaking space character

Closes #7252

Thanks to @efahl

(cherry picked from commit a2c21a62c9cd10b61baa48f4881cd8794930e4ff)