openwrt/luci #7260: luci-mod-network: Can't setup static route with "unspecified" interface.

bienias87, 2024-08-30:
## Steps to reproduce:
**Example: Replace the following lines and remove this**

1. go to: Network → routing
2. add new
3. try to set interface "unspecified"

## Actual behavior:
There is now option to set  "unspecified" interface. 
The interface we would like to use does not appear on the list, eg. zerotier. 

## Additional Information:
OpenWrt version [OpenWrt 23.05-SNAPSHOT (r24016-f791ec1f6d)]

If we set the route manually from the console, the display in Luci is correct. The interface is undefined, but the route works properly.

jow-, 2024-08-30:
The interface value is mandatory for uci declared routes. If your interface (zerotier) is not managed by uci, you cannot use uci route declarations for it.

bienias87, 2024-08-30:
No, you are wrong. Uci accepts a route without an interface, and that's how I set it.

For example:
#uci add network route
#uci set network.@route[-1].target='192.168.115.0'
#uci set network.@route[-1].netmask='255.255.255.0'
#uci set network.@route[-1].gateway='172.16.0.1'
#uci commit network

give us:
[image]

jow-, 2024-08-30:
Ok.

commit 648a0990 (2024-08-30): luci-mod-network: routes.js: make interface setting optional

Fixes: #7260
Ref: https://git.openwrt.org/5c2e5d5ea4e0dd9a0ed66154d818b96bf81664f4
closed 2024-08-30 by jow-