openwrt/luci #7261: "opkg update" fails to download packages

honeyreoss, 2024-08-30:
When attempting to run opkg update, I'm getting the following messages:

`opkg update
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/19.07.1/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
Downloading http://owrt.mooo.com/fw/openwrt/archerc5v4/packages/Packages.gz
*** Failed to download the package list from http://owrt.mooo.com/fw/openwrt/archerc5v4/packages/Packages.gz

Collected errors:
 * opkg_download: Failed to download http://owrt.mooo.com/fw/openwrt/archerc5v4/packages/Packages.gz, wget returned 8.`

How can i fix that? Change version openwrt?

jow-, 2024-08-30:
Remove references to `moo.com` from your opkg config and/or ask the supplier of your build for support. This is neither LuCI nor OpenWrt related.

closed 2024-08-30 by jow-

Saad-Imtiaz, 2024-10-18:
I had the same problem: and I was also unable to ping 1.1.1.1, which means the network was unreachable, so I tried the following fix, and it worked for me.
```
uci set network.lan.gateway="192.168.1.1"
uci set network.lan.dns="192.168.1.1"
uci commit
/etc/init.d/network restart
```

"192.168.1.1" needs to be replaced with the main routers IP Address.