openwrt/luci #7263: LUCI: Changing 'Devices' does not produce 'Interface has nn pending changes', leaving LUCI and /etc/config/network out of sync

The-QA-Geek, 2024-08-30:
## Steps to reproduce:

1. go to: Network → Interfaces
2. Edit the LAN Interface; make any change - for example, change the 'Maximum number of leased addresses'
3. Click 'Save'
4. In the 'LAN' row of LUCI, there will now be a message in blue saying 'Interface has 1 pending changes'
5. Click 'Save & Apply' and wait for LUCI to update

6. Click the 'Devices' tab
7. Edit the 'br-lan' device; remove 'lan3' and 'lan4' from the bridge
8. Click 'Save'

## Actual behavior:
- The message 'Interface has 1 pending changes' DOES NOT APPEAR
- The user is very likely to think that the changes have been saved, but they haven't - /etc/config/network still shows:

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'

## Expected behavior:
- LUCI should work the same way for Devices as for Interfaces - making a change and clicking 'Save' should produce a reminder that you have unsaved changes. 

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.4'
DISTRIB_REVISION='r24012-d8dd03c46f'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt 23.05.4 r24012-d8dd03c46f'
DISTRIB_TAINTS=''
```

jow-, 2024-08-31:
There's the universal unsaved change indicator in the upper right corner. There's no pending change message i nthe interface because no interface settings have been changed, Dives are different entities. There's no plan to change this behavior.

closed 2024-08-31 by jow-

The-QA-Geek, 2024-09-01:
I am well aware that:
 1) There's the universal unsaved change indicator in the upper right corner.
 2) Devices are different entities

This doesn't change the fact that this is poor UI design - if we teach the users to expect a change indicator in two places, then we should be consistent and make that happen for all changes. If we want to use only the universal unsaved change indicator in the upper right corner, then we should use ONLY that change indicator, and we should do that on EVERY screen.

Why do we ALWAYS put the 'Save' button at the bottom of the screen, on the right? Why not have some screens where it is at the top, or on the left? Keeping the UX consistent throughout is a hallmark of good UI design.

jow-, 2024-09-01:
The main reason for interfaces having this extraneous "has XX pending changes" indication is that it is possible to modify logical interfaces in a way that renders the complete interface preview row wrong. You could e.g. change the protocol of an interface from static to dhcp or assign a different device to it and the preview row would still continue to display "Protocol: Static address" while in reality the interface is configured for DHCP. To avoid confusion and to mitigate this discrepancy somewhat, the row in the interfaces overview does not display runtime info anymore as soon as there's any config changes, but a small link stating that there's pending config changes to imply that current runtime state does not match queued config changes.

For devices the problem is not as severe since the device overview grid does show almost no runtime information besides MTU and MAC. The MTU and MAC values are updated accordingly when the config is changed and modifications that would alter the entire nature of the device (such as changing the type from "network device" to "bridge" or similar) are not allowed.

So to recap:
- Interfaces show a generic placeholder "has XX changes" instead of state preview once any setting related to the interface is changed
- Devices properly reflect the modifications directly in the overview grid

The interface change indicator here is the actual outlier, all other gridsection based view with modifications in modal dialogs will not do this.

Without a PR implementing specific changes, this issue will not go anywhere, this is why I closed it.