openwrt/luci #7270: luci-app-statistics: Missing collectd module dependencies in downloads.openwrt.org package list

jivanpal, 2024-09-07:
## Steps to reproduce:

1. Install package `luci-app-statistics` on a fresh instance of OpenWrt 23.05.x from the OpenWrt repo (`https://downloads.openwrt.org/releases/23.05.x/packages/<arch>/luci`, which points to `https://downloads.openwrt.org/releases/packages-23.05/<arch>/luci`).
2. Go to [LuCI > Statistics > Setup > Network plugins], enable and configure the Conntrack and DNS plugins. Save & Apply changes.

## Actual behavior:

Watch the logs (`logread`) and see that collectd repeatedly fails to start due to the absence of the `dns` and `conntrack` modules:

```
<timestamp> daemon.err collectd[<num>]: configfile: stat (/etc/collectd/conf.d/*.conf) failed: No such file or directory
<timestamp> daemon.err collectd[<num>]: plugin_load: Could not find plugin "conntrack" in /usr/lib/collectd
<timestamp> daemon.err collectd[<num>]: plugin_load: plugin "cpu" successfully loaded.
<timestamp> daemon.err collectd[<num>]: plugin_load: Could not find plugin "dns" in /usr/lib/collectd
<timestamp> daemon.err collectd[<num>]: There is configuration for the `dns' plugin, but the plugin isn't loaded. Please check your configuration.
<timestamp> daemon.err collectd[<num>]: plugin_load: plugin "interface" successfully loaded.
<timestamp> daemon.err collectd[<num>]: plugin_load: plugin "load" successfully loaded.
<timestamp> daemon.err collectd[<num>]: plugin_load: plugin "memory" successfully loaded.
<timestamp> daemon.err collectd[<num>]: plugin_load: plugin "rrdtool" successfully loaded.
<timestamp> daemon.err collectd[<num>]: Error: Parsing the config file failed!
```

To fix this, the missing modules must be manually installed via the packages `collectd-mod-conntrack` and `collectd-mod-dns`.

## Expected behavior:

The LuCI app should work as configured immediately after installing just the `luci-app-statistics` package; collectd should start successfully without any additional configuration.

## Proposed fix

The packages `collectd-mod-conntrack` and `collectd-mod-dns` should be added as dependencies of `luci-app-statistics`, just like other necessary collectd modules currently are (namely: rrdtool, iwinfo, cpu, memory, interface, network).

## Additional Information:

OpenWrt version information from system `/etc/openwrt_release`:

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.4'
DISTRIB_REVISION='r24012-d8dd03c46f'
DISTRIB_TARGET='lantiq/xway'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt 23.05.4 r24012-d8dd03c46f'
DISTRIB_TAINTS=''
```

Ramon00, 2024-09-07:
Did you upgrade with existing config file? I am guessing it is failing because you installed and configured these in the past. I get the same myself (for different mods). I think what we could use is the UI showing which ones are missing/preventing it from starting. I dont think we can add the dependencies for everybody.

jivanpal, 2024-09-08:
Completely fresh install (my device was running out of RAM during sysupgrade, so I had to re-flash from stock). I don't remember if I had to manually install these when I was on 23.05.2 (which is what I upgraded from), but I expect so seeing as the package lists are one and the same.

Given that the LuCI app's features explicitly depend on the presence of these modules in order to function correctly, what would be the rationale for not including them?

hnyman, 2024-09-08:
There are dozens of possible collectd modules, which each may have X MB of dependencies. So, having somebody to accidentally turn on the one optional stat in LuCI stats config should not cause major installations.

So, WONTFIX.

closed 2024-09-08 by hnyman

jivanpal, 2024-09-08:
@hnyman Only certain modules are required by the LuCI app, not all of them. Why are some of these required modules listed as dependencies of luci-app-statistics, but not others?

hnyman, 2024-09-08:
Those are the (small) dependencies for the default plugin selection.

Ramon00, 2024-09-08:
As I mentioned, we could add to the luci app that it will show what dependencies to install instead of a user having to look in the log file (which I always end up doing after an upgrade)