openwrt/luci #7273: LuCI: Cannot set a static LAN address in the current WAN subnet

The-QA-Geek, 2024-09-09:
## Steps to reproduce:
1. go to: System -> Backup/Flash Firmware -> Perform Reset and reset the router to defaults
2. Determine the WAN subnet (the subnet provided by the upstream device, whether modem, router, whatever). Mine is using the 192.168.2.x subnet; my WAN interface is getting an address of 192.168.2.13, and the default gateway is 192.168.2.1. *I will assume these values for the rest of this discussion.*
3. Note the IP address of your PC; in my case it is 192.168.1.139.
4. Unplug the Ethernet cable from the WAN port
5. Go to Network -> Interfaces; click 'Edit' on the LAN interface
6. Change the LAN interface IPV4 address to 192.168.2.2
7. Click 'Save', then 'Save and Apply'
8. The warning screen pops up because we are changing the router's IP address; click “Apply and keep settings”. 
9. Open a new browser tab and browse to 192.168.2.2. As soon as the changes have been saved LuCI will appear - exactly as expected.
10. Now repeat steps 1 through 8 BUT SKIP Step 3 - leave the router connected to the upstream device that is handing out IP addresses in the 192.168.2.x subnet.
11. At Step 8, the PC will get a new IP address in the 192.168.2.x subnet, but the browser will never connect to the router at 192.168.2.2; the connection attempt will timeout. Pinging 192.168.2.2 will also fail.
12. In the browser, retry the connection to 192.168.2.2
13. Unplug the Ethernet cable from the WAN port - LuCI will immediately appear.

## Actual behavior:
1. The router successfully changes the DHCP server's address to the 192.168.2.x subnet; the PC gets a new address of 192.168.2.139.
2. The router is unable to change LuCI's IP address to any address in the WAN subnet while the WAN port is connected.
4. Removing the WAN connection resolves the problem and the router immediately begins working on the new IP address.

## Expected behavior:
```
OpenWRT should be able to change LuCI's IP address regardless of whether the WAN port is in the same subnet or not. If there is some deep architectural reason why this isn't easy, then LuCI should at least produce an informative message explaining the problem and the workaround. Perhaps it could just say "This change will require a reboot", and automatically reboot after making the change.
```

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.4'
DISTRIB_REVISION='r24012-d8dd03c46f'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt 23.05.4 r24012-d8dd03c46f'
DISTRIB_TAINTS=''
```

hnyman, 2024-09-10:
> OpenWRT should be able to change LuCI's IP address regardless of whether the WAN port is in the same subnet or not.

Well, if the router has the same subnet on both wan and lan, routing is impossible and traffic flows get confused. Connectivity from any side may be arbitrary if there is traffic from/to both sides.
There is no valid use case for having the same subnet on both sides unless you disable routing at the same time.

Regarding your steps 10-13, I would actually expect the attempt to timeout (due to the impossible routing), and the LuCI reverting the settings back to old ones after 45 seconds of waiting. And then you would have a new option to apply without checking that the new settings work. (no successful LuCI connection required during the timeout period)

jow-, 2024-09-10:
Configuration issue, nothing fixable on the LuCI end.

closed 2024-09-10 by jow-

The-QA-Geek, 2024-09-10:
@hnyman, @jow This is exactly the configuration described in the OpenWRT Wiki - https://openwrt.org/docs/guide-user/network/wifi/wifiextenders/bridgedap. The instructions there say (in part):

"
Go to Network → Interfaces and click on the Edit button of the LAN interface.
.
.
.
Assign the wireless AP an IP address.
By default, the main router will have an address of 192.168.1.1, so use 192.168.1.2 (or similar).
"

That page says 'Last modified: 2024/08/26 14:23by palebloodsky' - so just two weeks ago.

jow-, 2024-09-10:
Yes, the description is clearly wrong. Maybe it was supposed to say "so use 192.168.*2.1* (or similar)"