openwrt/luci #7276: luci-base: TypeError on every page

a1pha1337, 2024-09-12:
## Steps to reproduce:

1. Go to any page in LuCI

## Actual behavior:

1. Page loads with red error label with text: "TypeError_(...).format is not a function".  It fixes after several refreshes.

## Expected behavior:
**Example: Replace the following lines and remove this**

1. Page loads without any errors.

## Additional Information:
LuCI openwrt-23.05 branch (git-24.073.29889-cd7e519) / OpenWrt 23.05.3 (r23809-234f1a2efa)
OpenWrt version information from system `/etc/openwrt_release`
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.3'
DISTRIB_REVISION='r23809-234f1a2efa'
DISTRIB_TARGET='bcm27xx/bcm2711'
DISTRIB_ARCH='aarch64_cortex-a72'
DISTRIB_DESCRIPTION='OpenWrt 23.05.3 r23809-234f1a2efa'
DISTRIB_TAINTS=''
```

[image]
[image]

jow-, 2024-09-12:
`crypto.randomUUID()`,`page.js` and `injectedScript.bundle.js` do not belong to LuCI. I suspect a browser extension or some intermediate proxy is mangling the markup/scripts.

a1pha1337, 2024-09-12:
> `crypto.randomUUID()`,`page.js` and `injectedScript.bundle.js` do not belong to LuCI. I suspect a browser extension or some intermediate proxy is mangling the markup/scripts.
> 
> Some cursory internet research hints at a "MetaMask Chrome Extension" which commonly interferes with stuff...

Yeah, you are absolutely right, LuCI without extensions works perfectly. However, diasbling MetaMask extension didn't fix the problem. May be there is another extension that inteferes as well. Thank you!

closed 2024-09-12 by a1pha1337