openwrt/luci #7278: VLAN filtering checkbox does not reflect config file setting: vlan_filtering='1' shows checkbox empty.

vsessink, 2024-09-13:
## Steps to reproduce:

1. got to _Network_ → _Interfaces_ and click the _Devices_ tab
2. Click the _Configure_ button for `br-lan`
3. Click the _Bridge VLAN filtering_ tab
4. Click the _Enable VLAN filtering_ box
5. Click on the _[Save]_ button

Note that the UNSAVED CHANGES field now has 1 change: `uci set network.cfg030f15.vlan_filtering='1'`. So click _Save & Apply_ in order to apply the setting. 

## Actual behavior:

1. Device `br-lan` will now have `option vlan_filtering '1'`,
2. The Luci UI does not reflect that. Click the _Configure_ button for `br-lan` again, note that Enable VLAN filtering is _off_, i.e. not checked.

## Expected behavior:

The Luci UI shows that _Enable VLAN filtering_ is _on_, i.e. checked if `vlan_filtering` is `1`.

## Additional Information:
```
DISTRIB_ID='OpenWrt' DISTRIB_RELEASE='23.05.3' DISTRIB_REVISION='r23809-234f1a2efa'
DISTRIB_TARGET='mediatek/filogic' DISTRIB_ARCH='aarch64_cortex-a53' 
DISTRIB_DESCRIPTION='OpenWrt 23.05.3 r23809-234f1a2efa'
DISTRIB_TAINTS=''

```

systemcrash, 2024-09-14:
Does the change reflect in your underlying config?

vsessink, 2024-09-14:
Yes, it does, as stated under Actual Behavior: _Device br-lan will now have option vlan_filtering '1'_
So the configuration changes but the filtering checkbox stays empty as if no config change happened.

systemcrash, 2024-09-14:
The config... file. Check what it says in `/etc/config/network`

vsessink, 2024-09-14:
Sorry for - apparently - not being clear enough. Yes, the configuration file actually changes, but the checkbox doesn't reflect this change.

So to make it very clear, _before the change_ the `br-lan` entry in the config file `/etc/config/network` looks like:
```
config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0'
```
Then after checking the checkbox _enable vlan filtering_ in Luci, the `br-lan` config in the file `/etc/config/network` looks like:
```
config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0'
	option vlan_filtering '1'
```
But the _enable vlan filtering_ configuration option in Luci still shows an unchecked box.

systemcrash, 2024-09-14:
Then I imagine if you add some VLANs to that config, the setting makes sense. The code for this has:

```js
		o.updateDefaultValue = function(section_id) {
			var device = uci.get('network', s.section, 'name'),
			    uielem = this.getUIElement(section_id),
			    has_vlans = false;

			uci.sections('network', 'bridge-vlan', function(bvs) {
				has_vlans = has_vlans || (bvs.device == device);
			});

			this.default = has_vlans ? this.enabled : this.disabled;

			if (uielem && !uielem.isChanged())
				uielem.setValue(this.default);
		};
```

Try also adding a VLAN. What happens?

vsessink, 2024-09-14:
Yep, it switches on. Doesn't it do anything, besides setting a configuration option?