openwrt/luci #7280: luci-mod-status: Processes with large command words overlaps the display width

Waujito, 2024-09-14:
Hello!
I have a process that operates with a large string considered word by browsers and when the string gets larger and larger, the table starts to grow horizontally like on the screenshot
[image]

The fix for this will be 
```
.td[data-title="Command"] {
  word-break: break-word;
}
```

[image]

systemcrash, 2024-09-14:
Go ahead and propose a fix in a PR :)

Waujito, 2024-09-14:
I did it for only Processes table, because if I touch something like theme for all tables, it will break other stuff

commit dfb86709 (2024-09-18): luci-mod-status: break-word for large commands in processes

The commit fixes #7280
closed 2024-09-18 by systemcrash