openwrt/luci #7298: Feature request: ability to see active status and wake devices directly from "Active DHCP Leases" devices list (via WOL / Wake-on-LAN)

nekohayo, 2024-09-29:
Testing with the latest version of OpenWRT 23.05.5 r24106-10cc5fcd00 / LuCI openwrt-23.05 branch git-24.264.56413-c7a3562, I find myself missing a very nice convenience feature found in FreshTomato's web UI: the ability to directly wake devices from the overview status devices list.

In Tomato, it looks like this:

[image]

OpenWRT's Luci interface also has a devices listing table on the homepage (`admin/status/overview`), under the `Active DHCP Leases` and `Active DHCPv6 Leases` sections, with a very convenient "Set Static" button:

[image]

…so I'm thinking that, from a UX perspective, it could make a lot of sense to also have a convenience "Wake" button (with explanatory tooltip with something like `title="Click to wake up this device, if its network interface supports wake-on-LAN."` maybe) under an extra "WOL" or "Wake-on-LAN" column

I suspect this might be a technically simpler and more elegant way to solve #4188. It would also be more convenient for users, because there would be less clicks involved, it would be accessible in one click directly from the router's home page :)

systemcrash, 2024-10-07:
Good idea. There's `luci-app-wol`. That could be integrated into these device lists. Is this something you might want to undertake?

systemcrash, 2024-10-07:
There's the small matter of penalty, however: the `etherewake` tool requires 5kb (at least on my arm platform).

Neustradamus, 2024-12-19:
@systemcrash: Have you progressed on it?

nekohayo, 2024-12-22:
> Is this something you might want to undertake?

I wish, but I do not have the necessary programming (and networks) knowledge, so do not block on me; I am merely making the enhancement/UX suggestion, and am grateful to whoever may implement this :)

systemcrash, 2024-12-22:
One potential problem I find with this design is that once a device has gone to sleep, it disappears from the table because it no longer requests a lease. So... it kind of defeats the point. Thee work-around is to set very long lease times. A more robust WOL tool which caches all encountered MACs would be the way to do it. And we have one: `luci-app-wol`.

nekohayo, 2024-12-28:
Personally speaking, I think the default 24-hours lease time already makes this better than nothing, especially for the "waking up family members' computers that they use every day anyway" usecase; and I wouldn't mind telling the router to have multi-days lease times instead of 24-hours (especially if there was a button to drop/reset the lease, but I don't know if that's a possibility)