openwrt/luci #7299: luci-app-wifischedule - turns wifi off after power loss even if it is not time

Mile-Lile, 2024-09-30:
***

## Steps to reproduce:
**Example: Replace the following lines and remove this**

1. go to: Services→ WiFI Schedule → Enable Wifi Schedule → Enable logging → Enable BUSINESSHOURS
2. Enable all days → Start wifi 04:00 → Stop wifi 00:00 → Enable Force disabling wifi even if stations associated
3. Save & Apply
4. reboot router (simulate power off) in e.g. 17:00

## Actual behavior:
**Router starts with wifi off even if it is BUSINESSHOURS when wifi should be on**

## Expected behavior:
**Wifi should be on (regardless of rebooting), and should be off at the midhinght, script should check clock somehow**

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r27598-603a3c6955'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='ARMv8 Processor rev 4'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r27598-603a3c6955 / LuCI Master 24.264.56398~8449716'
ROUTER_MODEL='Xiaomi Mi Router AX3000T (OpenWrt U-Boot layout)'
```

Mile-Lile, 2024-09-30:
and these are errors from my logs, look at 19:19:44

[image]

Mile-Lile, 2024-10-01:
@systemcrash 
thx Paul... I reported here because link [here](https://openwrt.org/packages/pkgdata/luci-app-wifischedule) points to github luci, although it makes no sense that problem is in GUI... it is just simpe script that changes cron... how can I find the author of the script? 

I have ntp enabled... so probably script should wait execution after time sync....

hnyman, 2024-10-01:
> because link [here](https://openwrt.org/packages/pkgdata/luci-app-wifischedule) points to github luci, although it makes no sense that problem is in GUI

Well, you should look into the actual underlying functional package "wifischedule" instead of the GUI for it.

* https://openwrt.org/packages/pkgdata/wifischedule
* https://github.com/openwrt/packages/tree/master/net/wifischedule

@newkit seems to be the original author of the package, and @rindeal seems to have modified the startup functionality latest...
https://github.com/openwrt/packages/commits/3724047164218a58948933bf61eb089d36b67074/net/wifischedule

Mile-Lile, 2024-10-01:
@thx Hannu, will try to reach them..

Mile-Lile, 2024-10-01:
@newkit 
I opened new ticked as suggested, in [#25064](https://github.com/openwrt/packages/issues/25064) so I will close this one...

closed 2024-10-01 by Mile-Lile