openwrt/luci #7303: Suggestion for improvement im switch menu

siegfriedhoellrigl1, 2024-10-04:
It would be very nice if we could have a "Comment" field for each port under :
Network - Switch menu.

systemcrash, 2024-10-04:
What fields do you see today?

siegfriedhoellrigl1, 2024-10-04:
Today I can only make comments for the VLANs.
Not for the physical Ports.

dannil, 2024-10-04:
Which device are you running? The `Network` -> `Switch` menu isn't present on targets which has migrated to [DSA (Distributed Switch Architecture)](https://docs.kernel.org/networking/dsa/dsa.html), so if we want to have a place to make comments like this it should probably not be there.

siegfriedhoellrigl1, 2024-10-07:
At the moment I am using 3 TP-Links :
Archer A7 v5
TL-WDR4300 v1
TL-WR1043ND v2.
So you think, this is useless, because the switch menu will appear in a short time, correct ?

dannil, 2024-10-07:
> At the moment I am using 3 TP-Links : Archer A7 v5 TL-WDR4300 v1 TL-WR1043ND v2. So you think, this is useless, because the switch menu will appear in a short time, correct ?

Maybe not useless, some devices will probably never get DSA and stay on swconfig, so for them the switch menu would still appear. But, if we're thinking of a general solution about being able to add comments to the physical ports, there maybe could be a nice solution that would incorporate both swconfig and DSA targets.

systemcrash, 2024-10-07:
Admittedly, it's harmless to have a `comment` field, since it's only ever used in the GUI here. Lower level scripts aren't looking for it. DSA stuff is under Interfaces, right?

dannil, 2024-10-07:
> Admittedly, it's harmless to have a `comment` field, since it's only ever used in the GUI here. Lower level scripts aren't looking for it. DSA stuff is under Interfaces, right?

Yes, for example VLAN:s are configured as a new device under interfaces with DSA. If moving this to some "common" place (it would maybe even need a new UI) is too hard I'm fine with just adding a comment field as well.