openwrt/luci #7308: rp-pppoe-server. Openwrt 23.05. No menu services/rp-pppoe-server

Yrij-Zhavoronkov, 2024-10-06:
-Install rp-pppoe-server, luci-app-rp-pppoe-server, reboot.
-search for a new menu item. There is nothing.
-go to url: cgi-bin/luci/admin/services/rp-pppoe-server
`404 Not Found
Sorry, the object you requested was not found.

No page is registered at '/admin/services/rp-pppoe-server'. If this url belongs to an extension, make sure it is properly installed. If the extension was recently installed, try removing the /tmp/luci-indexcache file.

Unable to dispatch: /cgi-bin/luci/admin/services/rp-pppoe-server`

where is?

---
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.5'
DISTRIB_REVISION='r24106-10cc5fcd00'
DISTRIB_TARGET='mediatek/mt7622'
DISTRIB_ARCH='arm64'
DISTRIB_DESCRIPTION='LuCI openwrt-23.05 branch git-24.264.56413-c7a3562'
DISTRIB_TAINTS=''

systemcrash, 2024-10-07:
Closed by c344ad02a06a92fb9d8fab237cfc878a06b71ffd and 3aa60877ea4bc47b86c3ee11448db7503a20ae01

systemcrash, 2024-10-07:
You may need to wait a few days for the new package to roll out. Let me know how you get on with the new JS version.

closed 2024-10-07 by systemcrash