openwrt/luci #7310: convert outstanding LUA apps to JS

systemcrash, 2024-10-08:
Who is up for the task?

Key:

- [ ] **~~do not convert~~**
- [ ] 🗑️  _~~deprecate or remove~~_
- [ ] unassigned 
- [ ] not yet assigned with last known maintainer(s) tagged @octocat
- [x] assigned or accepted by @octocat 
- [x] ~~done~~ @octocat
- [x] ~~done~~ 

@stokito @dannil @Ramon00 ?

- [ ] luci-app-ahcp/luasrc @jow- 
- [ ] luci-app-clamav/luasrc  @ratkaj @lperkov 
- [ ] luci-app-coovachilli/luasrc @sbyx @jow- 
- [ ] luci-app-cshark/luasrc @lperkov 
- [ ] luci-app-dcwapd/luasrc @csonsino 
- [ ] luci-app-dnscrypt-proxy/luasrc @dibdot 
- [ ] luci-app-dockerman/luasrc @lisaac @feckert
- [ ] luci-app-dump1090/luasrc @Noltari 
- [ ] luci-app-dynapoint/luasrc @ascob
- [ ] luci-app-ltqtapi/luasrc @blogic 
- [ ] luci-app-lxc/luasrc @danrl 
- [ ] luci-app-mosquitto/luasrc @karlp
- [ ] luci-app-nft-qos/luasrc @rosysong
- [ ] luci-app-nut/luasrc @danielfdickinson
- [ ] luci-app-ocserv/luasrc @nmav
- [ ] luci-app-openvpn/luasrc @sbyx @jow- 
- [ ] luci-app-privoxy/luasrc @chris5560
- [ ] luci-app-radicale/luasrc @chris5560
- [ ] luci-app-radicale2/luasrc @danielfdickinson
- [ ] luci-app-siitwizard/luasrc @sbyx @jow- 
- [ ] luci-app-splash/luasrc @mmunz 
- [ ] luci-app-squid/luasrc  @ratkaj
- [ ] luci-app-tinyproxy/luasrc @EricLuehrsen
- [ ] luci-app-unbound/luasrc @EricLuehrsen 
- [ ] luci-app-vnstat/luasrc @jow- 
- [ ] luci-app-wifischedule/luasrc @newkit

Some of these might not make sense to convert. Some may have no users whatsoever - only download statistics will reveal info there, but those statistics are currently unavailable. At least a few of the maintainers likely have no time or desire to undertake such a conversion.

I think at least one app could be cut loose - siitwizard. Since we have native IPv4 over IPv6 now, and that's progressing nicely.

csonsino, 2024-10-08:
I'm not sure if the `dcwapd` feature ever had any users and I haven't looked at that project in a long time, but if I can get it up and running again I'll see if I can migrate the luci lua code.

Do you happen to have any useful links regarding this migration (developer notes, how-to's, etc)? My apologies, but I'm not very active in luci development and this is the first I'm hearing about lua being on the outs.

systemcrash, 2024-10-08:
Hi @csonsino 

That's not a bad idea. You could also review some of the recent conversions and do a side-by-side comparison ( e.g. luci-app-qos and luci-app-rp-pppoe-server ).

dannil, 2024-10-08:
I've started to convert `luci-app-squid` so you can assign me on that one.

stangri, 2024-10-08:
If unbound itself (the principal package) is still dependent on lua, maybe it's less of a priority to convert the luci-app-unbound from lua to javascript? Just my 0.02.

danrl, 2024-10-09:
Not sure how I ended up being the maintainer for `luci-app-lxc/luasrc` which I have never used. I used to co-maintain `luci-app-wiregaurd` and `wireguard`, but haven't been active for a while. Maybe a mix-up?

dhewg, 2024-10-09:
There's also luci-app-ddns. While the web frontend has been ported to 
JS, the underlying rpcd plugin has not.

To move away from lua, the rpcd lua plugin found at:
applications/luci-app-ddns/root/usr/libexec/rpcd/luci.ddns
needs to be ported to an rpcd ucode plugin.

Please add that to the list

systemcrash, 2024-10-09:
> If unbound itself (the principal package) is still dependent on lua, maybe it's less of a priority to convert the luci-app-unbound from lua to javascript? Just my 0.02.

What did you have in mind here @stangri ? I looked in the packages repo and unbound is a compiled C program, and a mish-mash of sh scripts. All the LUA appears to be MVC code on the luci side.

karlp, 2024-10-09:
I'm unlikely to have time for converting mosquitto in the near future.  I'm no longer employed on this, and I'd hoped to find enough free time to keep up with maintennance, but it's not really working out so far.  Seeing as paul donald has already _committed_ a change to this, maybe they can just take over that package :)

Ramon00, 2024-10-09:
I dont use any of these packages, but i had a quick look at luci-app-tinyproxy. I think I should be able to manage to translate that

stangri, 2024-10-09:
> What did you have in mind here @stangri ? I looked in the packages repo and unbound is a compiled C program, and a mish-mash of sh scripts. All the LUA appears to be MVC code on the luci side.

Oh, I might be wrong here and it's a kresd which is dependent on lua, I remember looking into the resolvers in the past and at least one of them was lua-dependent.

systemcrash, 2024-10-09:
> I dont use any of these packages, but i had a quick look at luci-app-tinyproxy. I think I should be able to manage to translate that

Shall I write you up for tinyproxy, @Ramon00 ?

systemcrash, 2024-10-09:
I think I will just remove cshark. It appears unmaintained. ( There's also the security aspect of dumping raw network frames to an external party who potentially has network breaches where this data gets exfiltrated. )

See github . com / cloudshark / cshark

Just run an ssh tap from wireshark. Infinitely more configurable. Great GUI with live view.

systemcrash, 2024-10-09:
> I'm not sure if the `dcwapd` feature ever had any users and I haven't looked at that project in a long time, but if I can get it up and running again I'll see if I can migrate the luci lua code.
> 

Perhaps @neheb has a better understanding of dcwapd? It looks like some feature that never really gained popularity. I can't tell if it requires special hardware. Is it like bonding two separate wifi channels into a single data-path for greater throughput or something?

https://github.com/ewsi/dcwapd
https://edgewaterwireless.com/dual-channel-wi-fi-tutorial/

neheb, 2024-10-09:
I have no idea what it is. It was placed in packages.

ping @csonsino

systemcrash, 2024-10-09:
ahcp doesn't seem like it's actively used. It reached a draft RFC about 14 years ago. And its origin page has broken links, and the most recent info indicates it is replaced by hnetd or shncpd, both of which are older and have a similar story.

https://www.irif.fr/~jch/software/ahcp/
https://github.com/jech/ahcpd

Looks like a candidate for the chop.

EricLuehrsen, 2024-10-10:
I don't know if I will be able to dig through `luci-app-unbound.` My time availability is different than it was. Notes: The zone lists are complex, but less clear documentation, are probably straight forward. The relationships to `odhcpd` and `dnsmasq` were implemented with some liberties in LUA environment that maybe are a problem here.

dannil, 2024-10-10:
> I think I will just remove cshark. It appears unmaintained. ( There's also the security aspect of dumping raw network frames to an external party who potentially has network breaches where this data [...]

If you're thinking of removing it, maybe it should be dropped from the packages repo first for the sake of completeness?

newkit, 2024-10-10:
I am not overly happy hearing about the move to JS given my limited time. What time frame are we talking about here?

For what it's worth - there must be some users of `luci-app-wifischedule` - at least there are some issues / PRs every now and then.

Ramon00, 2024-10-10:
> Shall I write you up for tinyproxy, @Ramon00 ?

 yeah go for it

dannil, 2024-10-10:
> I am not overly happy hearing about the move to JS given my limited time. What time frame are we talking about here?
> 
> For what it's worth - there must be some users of `luci-app-wifischedule` - at least there are some issues / PRs every now and then.

There's no set time fram but for what it's worth this has been a goal [since December 2019](https://github.com/openwrt/luci/issues/3378#issuecomment-560284715) so might as well get some traction on it, it's one of the pinned issues.

dibdot, 2024-10-12:
luci-app-dnscrypt-proxy should be removed, it's outdated and I haven't used it for years. 
Same applies to luci-app-lxc. If someone else wants to migrate this, then please do. I'm not going to do it.

itay-sho, 2024-10-15:
would love to give a shot on dockerman

systemcrash, 2024-10-15:
> would love to give a shot on dockerman

Go for it.

systemcrash, 2024-10-18:
> I'm not sure if the `dcwapd` feature ever had any users and I haven't looked at that project in a long time, but if I can get it up and running again I'll see if I can migrate the luci lua code.

@csonsino dcwapd is now converted (as of 82e5c888408a9728576f584088432289809fa293 ).

csonsino, 2024-10-18:
> @csonsino dcwapd is now converted (as of [82e5c88](https://github.com/openwrt/luci/commit/82e5c888408a9728576f584088432289809fa293) ).

Awesome, thank you @systemcrash! I had just gotten a system up and running with the old code so I'll give the new stuff a quick test.

systemcrash, 2024-10-23:
Any progress there @itay-sho ?

jow-, 2024-10-23:
I won't have time to deal with luci-app-openvpn, and given it's current state it is likely better to simply drop it. OpenVPN support should be done from scratch, using wireguard as role model and implemented as proto handler to better fit within the network config concept. Likely also with a focus on setting up a client with preexisting configs.

systemcrash, 2024-10-28:
So this effectively means openvpn also needs an overhaul to become a `proto` in the packages repo.

Is this what you had in mind?

Ramon00, 2024-10-28:
@newkit I had a quick look at luci-app-wifischedule looks pretty straightforward. Want me to convert it?

systemcrash, 2024-10-28:
We’ve not heard from @newkit since this thread started so fire away. It’s all yours @Ramon00

newkit, 2024-10-29:
@Ramon00 I‘d really appreciate that! 🤙🏼

Ramon00, 2024-10-29:
Alright :)

itay-sho, 2024-11-07:
> Any progress there @itay-sho ?

hey, sorry for disappearing. got a little bit busy.
I think it'll be better to take me off as the one in charge. In case I will find some time I will update here

systemcrash, 2024-11-25:
@feckert @hnyman are you aware of previous efforts to convert openvpn init stuff to a proto handler? I stumbled across a few but it seems nothing really made it in or was finished. openvpn seems to need a zillion variables.

dannil, 2024-11-28:
I took a look at `luci-app-dynapoint` and the upstream project seems dead at https://github.com/thuehn/DynaPoint, the last update to it in packages (which isn't treewide) is in May 2017 (https://github.com/openwrt/packages/commits/4695e2c10fb00e3ead4a30313a2db2de5be0dedd/net/dynapoint). So at least if we don't delete it we should most likely not spend any time on converting it.

cgnv, 2025-07-24:
[systemcrash](https://github.com/systemcrash) hi, i found a bug into [luci-proto-openconnect](https://github.com/openwrt/luci/commit/aa955d6465b4d0f00cc713904e2de7bfb0cbd062), shortly - it seems [_readfile](https://github.com/openwrt/luci/commit/aa955d6465b4d0f00cc713904e2de7bfb0cbd062#diff-5c19c1fca645536904f7a6a193289ea059d90621c54723cdb7dcddb88732a3cbR14-R21) should return false in case of errors, not error string (it affects following checks and ui)
Would it be helpful if I open an issue, or suggest a PR, or is the description sufficient as is?

systemcrash, 2026-02-04:
Wow. Only openvpn left to go.....

danielfdickinson, 2026-03-08:
@edrikk I haven't used it yet, but looking at the commit logs, I think openvpn has been switched to be a protocol (the same way WireGuard is configured). So you, I believe, create a network interface with protocol OpenVPN, which is managed by the networking subsystem, rather having a separate openvpn service and page.

edrikk, 2026-03-09:
You are right.  I think this might still need work.
For example, I see the names of the openvpn servers/clients but maybe as my setups are done using /etc/openvpn/proton.ovpn and /etc/openvpn/proton.auth, they are missing all the setups.

Ok, so I guess I have to reset them up -- I see there is a tab that lets you paste the .ovpn file in.  However, the contents there-in dont seem to be used to populate the other settings, and only a few settings are picked-up.

Save/Apply does not bring up the Interface as of course the settings are all incorrect/missing.