openwrt/luci #7314: module 'luci.ucodebridge' not found

TheSainEyereg, 2024-10-08:
## Steps to reproduce:
**Example: Replace the following lines and remove this**

1. Login to the LuCI

## Actual behavior:
Index page crashes (thanks god other pages are working)
```
module 'luci.ucodebridge' not found:
no field package.preload['luci.ucodebridge']
no file './luci/ucodebridge.lua'
no file '/usr/share/lua/luci/ucodebridge.lua'
no file '/usr/share/lua/luci/ucodebridge/init.lua'
no file '/usr/lib/lua/luci/ucodebridge.lua'
no file '/usr/lib/lua/luci/ucodebridge/init.lua'
no file './luci/ucodebridge.so'
no file '/usr/lib/lua/luci/ucodebridge.so'
no file '/usr/lib/lua/loadall.so'
no file './luci.so'
no file '/usr/lib/lua/luci.so'
no file '/usr/lib/lua/loadall.so'

In [anonymous function](), file [C]
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 133, byte 10:
  called from function [arrow function] (/usr/share/ucode/luci/runtime.uc:141:63)
  called from function render ([C])
  called from function [anonymous function] (/usr/share/ucode/luci/runtime.uc:141:64)
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:778:34)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:1003:48)
  called from anonymous function (/www/cgi-bin/luci:39:13)

 `            die(ex);`
  Near here --------^
```
## Expected behavior:
It should display index page

## Additional Information:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.4'
DISTRIB_REVISION='r24012-d8dd03c46f'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 23.05.4 r24012-d8dd03c46f'
DISTRIB_TAINTS=''
```

As I remember, I installed luci-app-openvpn, but as It was no longer needed I removed It. That's it

dannil, 2024-10-08:
Did you leave the `Automatically remove unused dependencies` checkbox checked? This is a known issue that shows up from time to time. 

You should be able to access LuCI again if you run `opkg install luci-lua-runtime`, see https://forum.openwrt.org/t/removed-clamav-package-with-all-dependacies-and-broke-luci/204187/4

TheSainEyereg, 2024-10-09:
Yep, now It's working. Thanks

closed 2024-10-09 by TheSainEyereg

nitinjs, 2024-10-16:
commenting for better reach
I got this error when I uninstalled `luci-app-ddns` keeping `Automatically remove unused dependencies` checked

systemcrash, 2024-10-16:
> Did you leave the `Automatically remove unused dependencies` checkbox checked? This is a known issue that shows up from time to time.
>
> You should be able to access LuCI again if you run `opkg [...]

It would seem that the problem is modules which perhaps do not properly list all their dependencies.