openwrt/luci #7315: luci-base:

systemcrash, 2024-10-08:
ping @jow- 

## Steps to reproduce:

1. go to: Syslog
2. observe `daemon.err` log lines like:

```
Tue Oct  8 18:10:07 2024 daemon.err uhttpd[2772]: [info] luci: accepted login on /admin/blah for root from x.x.x.x
```

This appears due to `dispatch.uc` which has:
```js
function syslog(prio, msg) {
	warn(sprintf("[%s] %s\n", prio, msg));
}
```

So we get `daemon.err` despite the message electing the `info` log class. Can this be improved?

## Expected behavior:

Correct log class for syslog messages from uhttpd.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.5'
DISTRIB_REVISION='r24106-10cc5fcd00'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 23.05.5 r24106-10cc5fcd00'
DISTRIB_TAINTS=''
```

jow-, 2024-10-08:
Make it `printf()` instead of `warn(sprintf())`. Should probably decide based on prio.

systemcrash, 2024-10-09:
That was one of the first things I tried but nothing is then emitted to the syslog. 😒

I thought you would know better what to try after that.

systemcrash, 2024-10-09:
I'll try printf, maybe I just tried sprintf.

systemcrash, 2024-10-09:
No - those don't seem to emit anything. 🤷

jow-, 2024-10-09:
Yeah, I recall. It's because procd captures uhttpd's stderr and relays it as daemon.err which is where LuCI's warn() output ends up. stdout is reserved for cgi communication. Not solvable unless you invest time implementing a full blown `syslog()` binding (some effort) or invoking the external logger util (overhead).