openwrt/luci #7326: Cannot read properties of null (reading 'remove')

jura43, 2024-10-13:
I fresh installed open-wrt on my Asus AX6000 router. What I configured in DHCP and DNS settings is deleted default instance and created new one with name work. After that I am not able to open DHCP and DNS settings.

## Steps to reproduce:
1. go to: Network → DHCP and DNS 

## Actual behavior:
[image]

## Additional Information:
```
root@OpenWrt:~# cat /etc/config/dhcp

config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '100'
        option leasetime '12h'
        option dhcpv4 'server'
        option dhcpv6 'server'
        option ra 'server'
        list ra_flags 'managed-config'
        list ra_flags 'other-config'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'

config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/tmp/hosts/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'
        option loglevel '4'

config dhcp 'Work'
        option start '100'
        option limit '200'
        option leasetime '12h'
        list server '1.1.1.1'
        option rebind_protection '0'
        option localservice '0'
        option interface 'Work'
        list dhcp_option '6,192.168.20.1'

config dhcp 'Guest'
        option interface 'Guest'
        option start '100'
        option limit '200'
        option leasetime '12h'
        list dhcp_option '6,1.1.1.1,1.0.0.1'
```

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r27751-f8c22c9bff'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r27751-f8c22c9bff'
DISTRIB_TAINTS=
```

jura43, 2024-10-13:
I think I am able to reproduce this.
I create interface, enable DHCP server, save. DHCP and DNS page still working. But if I edit DHCP settings in interface windows for example I change range of IP addresses and save. Then DHCP and DNS stop working.
Here is config before and after editing DHCP in interface menu.
```
config dnsmasq
        option domainneeded '1'
        option boguspriv '1'
        option filterwin2k '0'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option nonegcache '0'
        option cachesize '1000'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
        option nonwildcard '1'
        option localservice '1'
        option ednspacket_max '1232'
        option filter_aaaa '0'
        option filter_a '0'

config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option dhcpv4 'server'
        option dhcpv6 'server'
        option ra 'server'
        option ra_slaac '1'
        list ra_flags 'managed-config'
        list ra_flags 'other-config'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'

config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/tmp/hosts/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'
        option loglevel '4'

config dhcp 'guest'
        option interface 'guest'
        option start '100'
        option limit '150'
        option leasetime '12h'
```

```

config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option dhcpv4 'server'
        option dhcpv6 'server'
        option ra 'server'
        option ra_slaac '1'
        list ra_flags 'managed-config'
        list ra_flags 'other-config'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'

config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/tmp/hosts/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'
        option loglevel '4'

config dhcp 'guest'
        option interface 'guest'
        option start '100'
        option limit '200'
        option leasetime '12h'

config dhcp 'work'
        option start '100'
        option limit '200'
        option leasetime '12h'
        option rebind_protection '0'
        option localservice '0'
        option interface 'work'
        list server '1.1.1.1'
```

systemcrash, 2024-10-14:
Is there a trackback produced in the browser console?
I could not reproduce this.

jura43, 2024-10-14:
I manually added `config dnsmasq` block to dhcp file now I don't have any issues, and not able to reproduce it again. This might  seems to be happening only on fresh  install, since now I can edit DHCP range with issues.
I checked console, and I if remember correctly it was same message as in Web interface.

systemcrash, 2024-10-17:
At a guess, what's happening is the deletion of the first dnsmasq instance, which is anonymous. Anything added after that is a named instance. And dnsmasq handles those slightly differently. Maybe we shouldn't be able to delete the first instance.

ananthb, 2025-03-19:
I have the same issue on one of my routers. Didn't see it on two other OpenWRT fresh installs. Anything I can share to help?

burny0202, 2025-03-22:
Also have same issue - I think...

TypeError

nodes.querySelector(...) is null

vincejv, 2026-09-12:
Update as of Sept 2026...

The issue is reproducible on fresh installs by removing `dnsmasq` package and replacing it with `dnsmasq-full`... it's due to the missing `config dnsmasq` in `/etc/config/dhcp`