openwrt/luci #7327: luci-mod-rpc and home assistant

NA7KR, 2024-10-13:
Please can you help with this error:

2024-10-13 13:59:46.727 DEBUG (SyncWorker_1) [openwrt_luci_rpc.openwrt_luci_rpc] _call_json_rpc : https://192.168.10.1/cgi-bin/luci/rpc/auth
2024-10-13 13:59:46.729 DEBUG (SyncWorker_0) [openwrt_luci_rpc.openwrt_luci_rpc] _call_json_rpc : https://192.168.210.1/cgi-bin/luci/rpc/auth
2024-10-13 13:59:46.957 ERROR (MainThread) [homeassistant.components.device_tracker] Error setting up platform legacy luci
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/device_tracker/legacy.py", line 335, in async_setup_legacy
scanner = await hass.async_add_executor_job(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
result = self.fn(*self.args, **self.kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/luci/device_tracker.py", line 44, in get_scanner
scanner = LuciDeviceScanner(config[DEVICE_TRACKER_DOMAIN])
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/luci/device_tracker.py", line 55, in __init__
self.router = OpenWrtRpc(
^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/openwrt_luci_rpc/__init__.py", line 34, in __init__
self.router = OpenWrtLuciRPC(
^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/openwrt_luci_rpc/openwrt_luci_rpc.py", line 61, in __init__
self._refresh_token()
File "/usr/local/lib/python3.12/site-packages/openwrt_luci_rpc/openwrt_luci_rpc.py", line 69, in _refresh_token
self.token = self._call_json_rpc(auth_url, 'login',
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/openwrt_luci_rpc/openwrt_luci_rpc.py", line 261, in _call_json_rpc
raise LuciRpcUnknownError("Invalid response from luci: %s", res)
openwrt_luci_rpc.exceptions.LuciRpcUnknownError: ('Invalid response from luci: %s', <Response [500]>)
2024-10-13 13:59:47.179 INFO (SyncWorker_0) [openwrt_luci_rpc.openwrt_luci_rpc] Luci RPC login was successful

 OpenWrt 23.05.3, r23809-234f1a2efa
 -----------------------------------------------------
root@SalemOpenWrt:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.3'
DISTRIB_REVISION='r23809-234f1a2efa'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION

systemcrash, 2024-10-14:
In short: no. But in my personal experience with that plugin, it's twitchy from time to time, but in most release versions seems to work fine. Perhaps upgrade HA version.

closed 2024-10-14 by systemcrash

NA7KR, 2024-10-14:
I should be on the latest version