openwrt/luci #7328: luci-app-adblock-fast: implement pause button

stangri, 2024-10-14:
Hey @systemcrash @dannil, seeing your efforts to convert lua apps to js apps, I wonder if I could solicit your help with luci-app-adblock-fast.

The principal package has support for the `pause` function which (with one command as opposed to stop, then wait, then start) moves the resolver's blocklist file to a temp location, restarts the resolver, waits (from 1 to 60 seconds) then moves the block-list back and restarts the resolver again.

I've struggled to implement it in the luci app due to the time-outs. I have the [rpcd call implemented](https://github.com/stangri/luci-app-adblock-fast/blob/main/root/usr/libexec/rpcd/luci.adblock-fast#L14) but I don't know how to properly handle long waits for the return from the call in the luci app, any ideas?

Thanks!

systemcrash, 2024-10-14:
Curious: what's the necessity for the long running process? What does it ensure it achieves? Is it like a big flush operation?

dannil, 2024-10-15:
Just to clarify, is the `pause` function basically a way to temporarily disabling adblock as the blocklist is moved to somewhere else so the package just assumes to allow everything? Do you need a way then to act on when the timeout (the one from 1 to 60 seconds) completes without blocking the UI in the meantime?

stangri, 2024-10-16:
Thank you for your replies.

> Curious: what's the necessity for the long running process? What does it ensure it achieves? Is it like a big flush operation?

It may be desirable to wait for up to 60 seconds to make sure that the previously blocked domain name resolves, this, plus two resolver restarts require a long wait.

> Just to clarify, is the `pause` function basically a way to temporarily disabling adblock as the blocklist is moved to somewhere else so the package just assumes to allow everything? 

Yes, blocklist is moved and resolver is restarted to essentially remove the blocklist from operation. After the timeout the blocklist is placed again somewhere where resolver will pick it up and resolver is restarted again.

> Do you need a way then to act on when the timeout (the one from 1 to 60 seconds) completes without blocking the UI in the meantime?

I think it would be OK to block UI for the period of the time out and I can load the time out length variable so, ideally the countdown is displayed within the modal window.

If there's an option to block the UI for the duration of timeout (plus resolver restarts), the UI doesn't need to be refreshed afterwards.

systemcrash, 2024-10-17:
How about a sleep operation in a ucode script when handling these? It seems much better suited to run the operations on the local device.

stangri, 2024-10-17:
@systemcrash if I understood you correctly -- the rpcd code is not an issue, I already have it, my issue is how to handle the 60 seconds long request from the javascript/luci app.

systemcrash, 2024-10-19:
Yeah, I might eventually come up with a solution, but long running operations require the client to continually poll the HTTP endpoint. So maybe a 302 redirect would work. Is there a way to forcibly empty caches of the resovler so a more immediate restart would work?

systemcrash, 2024-10-20:
Sooo..... another idea might be to do what happens when the config is applied. We have a 90 second count-down until we get an "all-clear". Maybe utilise something like that?

Either co-opt the function or create your own?:

https://github.com/openwrt/luci/blob/cbdbdd579d7c388a48c1f57dc93d17d7bf15c829/modules/luci-base/ucode/controller/admin/uci.uc#L89-L117

stangri, 2024-10-20:
> Either co-opt the function or create your own?:
> 
> https://github.com/openwrt/luci/blob/cbdbdd579d7c388a48c1f57dc93d17d7bf15c829/modules/luci-base/ucode/controller/admin/uci.uc#L89-L117

Sorry, I may need some help understanding how can I use this. Isn't ucode the router-running piece of code? I do have the RPCD call working, I need help figuring out what type of javascript code needs to be implemented to support a long-running rpcd call.

systemcrash, 2024-10-20:
Does the 'pause' action have any kind of return or timeout? Reimplement the apply function above as a .uc script called like `pause_await`, and have this exported function call your rpc 'pause' function that returns after the necessary 60 seconds. Then your new e.g. `pause_await` routine responds after the allocated 'timeout'.

You might call this `pause_await` from the JS page by defining a `pause_await` function triggered by the `pause` button on the page. Compare with how e.g. `apply` is implemented here when you apply the config.

https://github.com/openwrt/luci/blob/cbdbdd579d7c388a48c1f57dc93d17d7bf15c829/modules/luci-base/htdocs/luci-static/resources/ui.js#L4719-L4781