openwrt/luci #7330: Missing ipproto keyword in network config rule section

TheRainstorm, 2024-10-17:
ip-rule[^1] supports `ipproto` keyword, so we can routing UDP traffic in different manner using following command:

```shell
ip ru add from all ipproto udp table udp_table prior 10
```

Openwrt `ip` util also support this keyword. Howerver, In Luci we cannot setting the `ipproto` keyword. The network configuration file also doesn't support setting this keyword. [^2]

In order to implement routing according to L4 protocol, we can only use command line or routing using fwmark with the help of nftables. It will be more convient if the `ipproto` is added.

## Steps to reproduce:

1. go to: Network → Routing→ IPv4 rules
2. no method to set `ipproto`

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.0'
DISTRIB_REVISION='r23497-6637af95aa'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 23.05.0 r23497-6637af95aa'
DISTRIB_TAINTS=''
```

I also installed `ip-full`

[^1]: https://man7.org/linux/man-pages/man8/ip-rule.8.html
[^2]: https://openwrt.org/docs/guide-user/network/routing/ip_rules

systemcrash, 2024-10-17:
This is not complex to add within luci, but luci depends on what exists in the openwrt base system. I sent a patch for netifd to the mailing list and opened a PR which will probably take about 17 years to get acknowledged.

See https://github.com/openwrt/netifd/pull/36

systemcrash, 2024-11-25:
Step 1 complete - change committed to netifd. 
Step 2: update openwrt to use newer netifd.

systemcrash, 2024-12-13:
Closed by a6e3ea42ac9bbf7ba84f4f515c2c2986b172d234

closed 2024-12-13 by systemcrash