openwrt/luci #7334: openwrt_luci_rpc hrlp please

NA7KR, 2024-10-18:
Masquerade MAC> IP and NAMES: 
2024-10-18 08:58:36.635 DEBUG (SyncWorker_2) [openwrt_luci_rpc.openwrt_luci_rpc] Checking for connected devices 
2024-10-18 08:58:36.635 DEBUG (SyncWorker_2) [openwrt_luci_rpc.openwrt_luci_rpc] _call_json_rpc : https://XXX.XXX.XXX.XXX/cgi-bin/luci/rpc/ip?auth=XXXXXXXXXXXXXXXXXXXXXXXXXXXX
2024-10-18 08:58:36.975 DEBUG (SyncWorker_2) [c] _call_json_rpc : https://XXX.XXX.XXX.XXX/cgi-bin/luci/rpc/uci?auth=XXXXXXXXXXXXXXXXXXXXXXXXXXXX
2024-10-18 08:58:37.102 DEBUG (SyncWorker_2) [openwrt_luci_rpc.utilities] DNS name lookup for mac MAC1 found HOST1
2024-10-18 08:58:37.102 DEBUG (SyncWorker_2) [openwrt_luci_rpc.utilities] DNS name lookup for mac MAC2 found HOST2
2024-10-18 08:58:37.102 DEBUG (SyncWorker_2) [openwrt_luci_rpc.utilities] DNS name lookup for mac MAC3 found HOST3
2024-10-18 08:58:37.103 DEBUG (SyncWorker_2) [openwrt_luci_rpc.utilities] DNS name lookup for mac MAC4 found HOST4
2024-10-18 08:58:37.106 DEBUG (SyncWorker_2) [openwrt_luci_rpc.utilities] DNS name lookup for mac MAC5 found HOST5
2024-10-18 08:58:37.106 DEBUG (SyncWorker_2) [openwrt_luci_rpc.utilities] DNS name lookup for mac MAC6 found MAC6
2024-10-18 08:58:37.107 DEBUG (SyncWorker_2) [openwrt_luci_rpc.utilities] DNS name lookup for mac MAC7 found HOST7
2024-10-18 08:58:37.107 DEBUG (SyncWorker_2) [openwrt_luci_rpc.openwrt_luci_rpc] [Device(dev='br-lan.211', stale=False, mac='MAC1', noarp=False, permanent=False, failed=False, family=4, proxy=False, router=False, reachable=True, probe=False, delay=False, incomplete=False, ip='IP1', hostname='MAC1', host='XXX.XXX.XXX.XXX'), 
Device(dev='br-lan.210', stale=False, mac='MAC2', noarp=False, permanent=False, failed=False, family=4, proxy=False, router=False, reachable=True, probe=False, delay=False, incomplete=False, ip='IP2', hostname='HOST1', host='XXX.XXX.XXX.XXX'), 
Device(dev='wan', stale=False, mac='MAC8', noarp=False, permanent=False, failed=False, family=4, proxy=False, router=False, reachable=True, probe=False, delay=False, incomplete=False, ip='IP3', hostname=None, host='XXX.XXX.XXX.XXX'), 
Device(dev='br-lan.211', stale=False, mac='MAC6', noarp=False, permanent=False, failed=False, family=4, proxy=False, router=False, reachable=True, probe=False, delay=False, incomplete=False, ip='IP4', hostname='MAC6', host='XXX.XXX.XXX.XXX'), 
Device(dev='br-lan.211', stale=False, mac='MAC9', noarp=False, permanent=False, failed=False, family=4, proxy=False, router=False, reachable=True, probe=False, delay=False, incomplete=False, ip='IP5', hostname=None, host='XXX.XXX.XXX.XXX'), 
Device(dev='br-lan.213', stale=False, mac='MAC7', noarp=False, permanent=False, failed=False, family=4, proxy=False, router=False, reachable=True, probe=False, delay=False, incomplete=False, ip='IP6', hostname='HOST7', host='XXX.XXX.XXX.XXX'), 
Device(dev='br-lan.211', stale=False, mac='MAC10', noarp=False, permanent=False, failed=False, family=4, proxy=False, router=False, reachable=True, probe=False, delay=False, incomplete=False, ip='IP7', hostname=None, host='XXX.XXX.XXX.XXX')]
2024-10-18 08:58:37.108 DEBUG (SyncWorker_2) [homeassistant.components.luci.device_tracker] Luci get_all_connected_devices returned: 
[Device(dev='br-lan.211', stale=False, mac='MAC1', noarp=False, permanent=False, failed=False, family=4, proxy=False, router=False, reachable=True, probe=False, delay=False, incomplete=False, ip='IP1', hostname='MAC1', host='XXX.XXX.XXX.XXX'), 
Device(dev='br-lan.210', stale=False, mac='MAC2', noarp=False, permanent=False, failed=False, family=4, proxy=False, router=False, reachable=True, probe=False, delay=False, incomplete=False, ip='IP2', hostname='HOST1', host='XXX.XXX.XXX.XXX'), 
Device(dev='wan', stale=False, mac='MAC8', noarp=False, permanent=False, failed=False, family=4, proxy=False, router=False, reachable=True, probe=False, delay=False, incomplete=False, ip='IP3', hostname=None, host='XXX.XXX.XXX.XXX'), 
Device(dev='br-lan.211', stale=False, mac='MAC6', noarp=False, permanent=False, failed=False, family=4, proxy=False, router=False, reachable=True, probe=False, delay=False, incomplete=False, ip='IP4', hostname='MAC6', host='XXX.XXX.XXX.XXX'), 
Device(dev='br-lan.211', stale=False, mac='MAC9', noarp=False, permanent=False, failed=False, family=4, proxy=False, router=False, reachable=True, probe=False, delay=False, incomplete=False, ip='IP5', hostname=None, host='XXX.XXX.XXX.XXX'), 
Device(dev='br-lan.213', stale=False, mac='MAC7', noarp=False, permanent=False, failed=False, family=4, proxy=False, router=False, reachable=True, probe=False, delay=False, incomplete=False, ip='IP6', hostname='HOST7', host='XXX.XXX.XXX.XXX'), 
Device(dev='br-lan.211', stale=False, mac='MAC10', noarp=False, permanent=False, failed=False, family=4, proxy=False, router=False, reachable=True, probe=False, delay=False, incomplete=False, ip='IP7', hostname=None, host='XXX.XXX.XXX.XXX')]
Connected
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.3'
DISTRIB_REVISION='r23809-234f1a2efa'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt 23.05.3 r23809-234f1a2efa'
DISTRIB_TAINTS=''

systemcrash, 2024-10-19:
?

NA7KR, 2024-10-19:
Does not work this is error message above

systemcrash, 2024-10-19:
I've no idea what you're trying to do, nor what the problem or error is. Better to ask in HA forums.

closed 2024-10-19 by systemcrash