openwrt/luci #7339: luci-proto-wireguard: incorrect peer name detection when more than one peer with the same public key exist

this-username-has-been-taken, 2024-10-22:
If an OpenWrt device is used as a Wireguard client (i.e. no "listen port" is defined for the interface) it is allowed to define more than one peer with the same public key. In such case the peer name in `Status → WireGuard` menu might not be detected correctly because the code will only compare the public key of a peer and take the first match. To fix the issue it is required to check not only the public key of a peer, but also its `endpoint_host` and `endpoint_port`.
***
## Possible solution:
Replace line 79 in `luci/protocols/luci-proto-wireguard/root/usr/share/rpcd/ucode/luci.wireguard` file:
```
if (s.public_key == record[1])
```
with the following:
```
if (s.public_key == record[1] && s.endpoint_host + ":" + s.endpoint_port == record[3])
```
The solution is tested and works if the `endpoint_host` is an IP address. I didn't test it if the `endpoint_host` is an FQDN-name.

## Steps to reproduce:

1. Go to: Network → Interfaces → add a new WireGuard interface. The interface should not contain `listen port`, i.e. act as a client. Fill all other parameters (`private key`, `ip address`) with correct values.
2. Add a fake peer on the `Peers` tab with name `Incorrect peer 1`; set `Disabled` property `true`; fill `Public key` with a real public key value (see further); endpoint host and port with random values.
3. Optional. Add another fake Peer with the same parameters, but different name, e.x. `Incorrect peer 2`.
4. Add a real peer with the same public key (i.e. all peers must have the same public key from the real one) with name `Correct peer`; set `Disabled` property `false`; fill endpoint host and port with correct values.
5. The order of peers is critical, i.e. create fake peers first and the real one last.
6. Save and apply.
7. Set the new Wireguard interface up and running (i.e. connect to the real peer).
8. Go to `Status → WireGuard` page.
9. Observe the issue: the page will show Peer: `Incorrect peer 1` and all other parameters from the real peer.

## Actual behavior:
The `Status → WireGuard` page shows incorrect peer name with all other correct parameters.

## Expected behavior:
The `Status → WireGuard` page shows correct peer name and all other correct parameters.

## Additional Information:
Doesn't matter, but here it is:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r27859-f9da81d32f'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r27859-f9da81d32f'
DISTRIB_TAINTS='no-all'

```

systemcrash, 2024-10-22:
I see. Go ahead and make a PR. Would be great to test also with an FQDN. 

Could you illustrate this problem with a few screenshots? One of the status page and one of the peers tab?

this-username-has-been-taken, 2024-10-22:
I have checked: unfortunately the suggested solution doesn't work with the FQDN because `wg show all dump` command shows resolved IP address, but config stores the FQDN.
An additional step should be taken: convert FQDN to an IP address. And that is another challenge: there might be defined more that one IP for a FQDN. I will take a look into it.

Regarding screenshots. I am using modified version of the Wireguard protocol - it it OK? It is absolutely the same in terms of defining the peer name.

systemcrash, 2024-10-22:
Yeah, if an FQDN is backed by multiple A or AAAA records, or even SRV, there are no guarantees on which IP you get.

systemcrash, 2024-10-22:
Does something like the following work? Consider it pseudocode - I'm not sure which properties are needed right now
```js
if (s.public_key == record[1] && ((s.endpoint_host + ":" + s.endpoint_port == record[3]) || (s.endpoint_fqdn + ":" + s.endpoint_port == record[3])))
```

this-username-has-been-taken, 2024-10-22:
> Does something like the following work? Consider it pseudocode - I'm not sure which properties are needed right now
> 
> ```js
> if (s.public_key == record[1] && ((s.endpoint_host + ":" + s.endpoint_port == record[3]) || (s.endpoint_fqdn + ":" + s.endpoint_port == record[3])))
> ```

Nope, this won't work because `record[3]` is always an IP address (previously resolved if needed).

> Yeah, if an FQDN is backed by multiple A or AAAA records, or even SRV, there are no guarantees on which IP you get.

I believe a function with a loop is required to check all IPs

systemcrash, 2024-10-22:
> I believe a function with a loop is required to check all IPs.
> Unfortunately I am not that familiar with `ucode` language. I don't even know how to debug it or get output from a function to check it. I will still look into it though.

Largely synonymous with older JS, but some newer syntax is allowed. So

```js
for (var i = Things.length - 1; i >= 0; i--) {
	Things[i]
}
```
or
```js
for (var i = 0; i < Things.length; i++) {
	Things[i]
}
```

systemcrash, 2024-10-22:
Although there may be other ways to tackle this. A couple of screen-shots would be good.

this-username-has-been-taken, 2024-10-22:
> Although there may be other ways to tackle this. A couple of screen-shots would be good.

On the `Peers` tab I have defined two peers:

1. The first one is disabled and has an IP address as the endpoint.
2. The second one is enabled and has a FQDN as the endpoint.
3. These two peers have the same public key.
I have redacted some information due to privacy reasons.
[image]

On the `Status → WireGuard` page we can see that the peer name is incorrect, but all other data is correct (belongs to the `Correct peer 1`) including it's IP which has been resolved from the FQDN. I beg my pardon, but I had to redact private information.
[image]

So the problem lies only in finding the correct name of the peer.

systemcrash, 2024-10-22:
I see. Great. So the uci script identifies the incorrect peer name, since both entries share the same key. ( They're the same endpoint ). The output here depends on `wg show all dump`. This won't be impossible, but it seems like you'd need to reinvent the wheel with DNS to figure out the IPs to cover all cases.

this-username-has-been-taken, 2024-10-22:
What I thought was:
1. Check if we got an IP or FQDN from the LuCi config.
2. If IP - then concatenate it with port and simply compare with what we got form the `wg show all dump`.
3. If FQDN - then run `nslookup {FQDN goes here} | awk '/^Address: / { print $2 }'`
4. Loop through all received IPs
5. Concatenate each one with port and compare with what we got form the `wg show all dump`.

Because I don't know `ucode` well, I can't get results of step 3 and can't loop through them.

this-username-has-been-taken, 2024-10-22:
I came up with the following solution:
1. Added a new function to the `luci/protocols/luci-proto-wireguard/root/usr/share/rpcd/ucode/luci.wireguard` file:
```
function checkPeerHost(configHost, configPort, wgHost) {
		if (match(configHost, /^((25[0-5]|(2[0-4]|1\d|[1-9]|)\d)\.?\b){4}$/)) {
				return configHost + ":" + configPort == wgHost;
		} else {
				const ips = popen(`nslookup ${configHost} | awk '/^Address: / { print $2 }' 2>/dev/null`);
				if (ips) {
						for (let line = ips.read('line'); length(line); line = ips.read('line')) {
								const ip =  rtrim(line, '\n');
								if (ip + ":" + configPort == wgHost) {
										return true;
								}
						}
				}
		}
		return false;
}
```
2. Modified line 79:
```
if (s.public_key == record[1] && checkPeerHost(s.endpoint_host, s.endpoint_port, record[3]))
```

And... It works! Both with IPs ans FQDNs!

systemcrash, 2024-10-23:
Hmm - I wonder whether this process is simplified by using that tiny ip lookup package. It's tiny, so I guess it'd be ok:

https://openwrt.org/packages/pkgdata/resolveip

this-username-has-been-taken, 2024-10-23:
I liked it: no need to check if we got an IP and no need to parse the response. The code got simplified:

```
function checkPeerHost(configHost, configPort, wgHost) {
        const ips = popen(`resolveip ${configHost} 2>/dev/null`);
        if (ips) {
                for (let line = ips.read('line'); length(line); line = ips.read('line')) {
                        const ip =  rtrim(line, '\n');
                        if (ip + ":" + configPort == wgHost) {
                                return true;
                                                        }
                        }
                }
        return false;
}
```

I also added a check if a peer is disabled:
```
if (!s.disabled && s.public_key == record[1] && checkPeerHost(s.endpoint_host, s.endpoint_port, record[3]))
```

systemcrash, 2024-10-23:
Closed by 7acea818735a9c5e8577d6b57f359996956f15e4

systemcrash, 2024-10-23:
Thanks for the quick fix @this-username-has-been-taken. Good work!

closed 2024-10-23 by systemcrash

this-username-has-been-taken, 2024-10-23:
Thank you for all the advices and help! :)