openwrt/luci #7347: luci-proto-wireguard: incorrect peer name detection when devise acts as a server

this-username-has-been-taken, 2024-10-24:
Unfortunately a previously fixed bug (see https://github.com/openwrt/luci/issues/7339) caused another issue: if an OpenWrt device is used as a Wireguard server (i.e. "listen port" is defined for the interface) then peer names on the `Status → WireGuard` page are not detected at all: `Untitled peer` is shown instead of the real names.

***
## Suggested solution:
Technically it is possible to save more than one peer with the same public/private key pair on the `Peers` tab, but after the interface is restarted only one peer will be active (the latter one). All other peers with the same public/private key pairs will be ignored and not shown in the `wg show all dump` command output. Thus we do not need to check peer addresses in this case - it is enough to only check if a public key matches.
In the view of the above to fix the issue I suggest to modify line 92 in `luci/protocols/luci-proto-wireguard/root/usr/share/rpcd/ucode/luci.wireguard` file from:
```
if (!s.disabled && s.public_key == record[1] && checkPeerHost(s.endpoint_host, s.endpoint_port, record[3]))
```
to:
```
if (!s.disabled && s.public_key == record[1] && (!s.endpoint_host || checkPeerHost(s.endpoint_host, s.endpoint_port, record[3])))
```
I.e. do not check a peer's host if no endpoint is defined (interface acts as a server).

## Steps to reproduce:

1. go to: Network → Interfaces → create a Wireguard interface. The interface should contain `listen port`, i.e. act as a server. Fill all other parameters (private key, ip address, etc) with correct values.
2. Add one or more peers on the `Peers` tab. None of the peers should have `endpoint host` defined.
3. Save & Apply.
4. Restart the interface.
5. Navigate to `Status → WireGuard` page.
6. Observe the issue:
[image]

## Actual behavior:
All the peers on the `Status → WireGuard` page have `Untitled peer` name.

## Expected behavior:
All the peers on the `Status → WireGuard` page have correct names.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

Doesn't matter, but here it is:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r27902-6835ff8cbc'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r27902-6835ff8cbc'
DISTRIB_TAINTS='no-all'
'
```

closed 2024-10-24 by systemcrash

this-username-has-been-taken, 2024-10-24:
Thank you for moving the fix to the repo!

systemcrash, 2024-10-24:
No problem. Thanks for the prompt fix.

commit 760ac165 (2024-10-24): luci-proto-wireguard: follow-up fix for status overview

follow-up to 7acea818735a9c5e8577d6b57f359996956f15e4

Closes #7347
commit 07ca6037 (2024-10-25): luci-proto-wireguard: fixed bug with incorrect peer name detection

Fixed bug with incorrect peer name detection on `Status -> WireGuard`
page when more than one peer with the same public key exist:
1. Peers are now tested not only by public key, but also by
enabled/disabled status, peer host (both IP and FQDN are supported)
and port.
2. Added required `resolveip` dependency.

Closes #7342

(cherry picked from commit 7acea818735a9c5e8577d6b57f359996956f15e4)

luci-proto-wireguard: follow-up fix for status overview

follow-up to 7acea818735a9c5e8577d6b57f359996956f15e4

Closes #7347

(cherry pi [... cut]