openwrt/luci #7349: LuCI index page gives error before login (can not screen 'luaindex.htm' page)

evanzing, 2024-10-25:
on http://192.168.2.254/cgi-bin/luci/ page
cannot screen the home page  instead of this code :

Error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/template.lua:181: Failed to execute template 'themes/opentopd/header'.
A runtime error occurred: /usr/lib/lua/luci/dispatcher.lua:68: attempt to index global '__entries' (a nil value)
stack traceback:
/usr/lib/lua/luci/dispatcher.lua:68: in function 'node'
[string "/usr/lib/lua/luci/view/themes/opentopd/head..."]:19: in main chunk

In error(), file [C]
called from function run (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:37)
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 133, byte 10:
  called from function [arrow function] (/usr/share/ucode/luci/runtime.uc:141:63)
  called from function render ([C])
  called from function [anonymous function] (/usr/share/ucode/luci/runtime.uc:141:64)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:941:44)
  called from anonymous function (/www/cgi-bin/luci:39:13)

 `            die(ex);`
  Near here --------^

I made the image my self with ubuntu system

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='R27902-6835ff8cbc'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r27902-6835ff8cbc'
DISTRIB_TAINTS='no-all override'

evanzing, 2024-10-25:
[image]

hnyman, 2024-10-25:
There is no official theme "opentopd".
Likely the third-party theme that you have obtained from somewhere, is faulty.

closed 2024-10-25 by hnyman