openwrt/luci #7350: luci can't be installed with todays NIGHTLY openWRT 24-25th October 2024

bittin, 2024-10-25:
see log: 

BusyBox v1.36.1 (2024-10-24 18:17:27 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r27903-27657050d0
 -----------------------------------------------------
root@rainbowdash:~# opkg install luci
Installing luci (23.051.66410~a505bb1) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/luci/luci_23.051.66410~a505bb1_all.ipk
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.57~7fe213948abdaa7a70bd08d692784bd7-r1) for kmod-nf-reject
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.57~7fe213948abdaa7a70bd08d692784bd7-r1) for kmod-nf-reject6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.57~7fe213948abdaa7a70bd08d692784bd7-r1) for kmod-nf-log
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.57~7fe213948abdaa7a70bd08d692784bd7-r1) for kmod-nf-log6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.57~7fe213948abdaa7a70bd08d692784bd7-r1) for kmod-crypto-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.57~7fe213948abdaa7a70bd08d692784bd7-r1) for kmod-nft-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.57~7fe213948abdaa7a70bd08d692784bd7-r1) for kmod-nf-flow
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.57~7fe213948abdaa7a70bd08d692784bd7-r1) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.57~7fe213948abdaa7a70bd08d692784bd7-r1) for kmod-ipt-core
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/luci/luci_23.051.66410~a505bb1_all.ipk, wget returned 8.
 * opkg_install_pkg: Failed to download luci. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package luci.
root@rainbowdash:~# opkg update
Downloading https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/kmods/6.6.58-1-7fe213948abdaa7a70bd08d692784bd7/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_kmods
Downloading https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/kmods/6.6.58-1-7fe213948abdaa7a70bd08d692784bd7/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/telephony/Packages.sig
Signature check passed.
root@rainbowdash:~# opkg install luci
Installing luci (23.051.66410~a505bb1) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/luci/luci_23.051.66410~a505bb1_all.ipk
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.57~7fe213948abdaa7a70bd08d692784bd7-r1) for kmod-nf-reject
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.57~7fe213948abdaa7a70bd08d692784bd7-r1) for kmod-nf-reject6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.57~7fe213948abdaa7a70bd08d692784bd7-r1) for kmod-nf-log
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.57~7fe213948abdaa7a70bd08d692784bd7-r1) for kmod-nf-log6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.57~7fe213948abdaa7a70bd08d692784bd7-r1) for kmod-crypto-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.57~7fe213948abdaa7a70bd08d692784bd7-r1) for kmod-nft-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.57~7fe213948abdaa7a70bd08d692784bd7-r1) for kmod-nf-flow
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.57~7fe213948abdaa7a70bd08d692784bd7-r1) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.57~7fe213948abdaa7a70bd08d692784bd7-r1) for kmod-ipt-core
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/luci/luci_23.051.66410~a505bb1_all.ipk, wget returned 8.
 * opkg_install_pkg: Failed to download luci. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package luci.

hnyman, 2024-10-25:
It is a temporary issue related to kernel updates in the main/master snapshot builds and the fact that we have a two-step buildbot. The packages buildbot needs to update the packages with the SDK build including the updated kernel.

closed 2024-10-25 by hnyman

bittin, 2024-10-26:
@hnyman this is now fixed thank :)