openwrt/luci #7352: No “software”on ”system”

evanzing, 2024-10-25:
How should I do for this？
When I typef  this “opkg update&opkg install luci-app-opkg”
It told me that it was a unknow 
[image]

[image]

hnyman, 2024-10-25:
There are no "small" or "kenzo"directories in the normal download serves, so I have no idea, what firmware you are running.
luci-app-opkg should in x86_64/base directory, but has been renamed a few days ago to luci-app-package-manager

http://downloads.openwrt.org/snapshots/packages/x86_64/luci/luci-app-package-manager_24.297.79519~bcd13b9_all.ipk

this-username-has-been-taken, 2024-10-25:
`luci-app-opkg` has been renamed to `luci-app-package-manager`: https://github.com/openwrt/luci/issues/7345
Try running `opkg install luci-app-package-manager` instead.

systemcrash, 2024-10-25:
> `luci-app-opkg` has been renamed to `luci-app-package-manager`: [bcd13b9](https://github.com/openwrt/luci/commit/bcd13b918e2f30b8d19027a06e3d773a1b0ec4c0) Try running `opkg install luci-app-package-manager` instead.

Solution 👍

closed 2024-10-25 by systemcrash

hnyman, 2024-10-25:
Probably due to that luci-light thing that in practice removed opkg app from default LuCI in private builds...