openwrt/luci #7354: luci-proto-wireguard: suggestion to make peers sortable (enable drag and drop)

this-username-has-been-taken, 2024-10-25:
Hello!
If a Wireguard interface acts as a server (i.e. has a `listen port` defined), it usually has more than one peer defined. In such cases it might be very convenient to change their order, especially when adding a new peer: put it in "its place".
My suggestion is to enable `sortable` option for the Peers `TabSection`, i.e. allow users to change peers order on the corresponding tab.
***

## Suggested solution
To enable sorting on the Peers tab, it is required to set just one property for the tab, i.e. add one line after line 207 in `luci/protocols/luci-proto-wireguard/htdocs/luci-static/resources/protocol/wireguard.js` script:
```
ss.sortable = true;
```
I have added this line on the live OpenWrt system, restarted the webserver and rpcd daemons and it worked just fine (see screenshot below).
In fact writing this task took more time then applying the modification :)
[image]
After applying the modification it is now possible to rearrange `Peer 3` and `Peer 4` via web interface and make those guys with OCD happy!

## Steps to reproduce:
1. Go to: Network → Interfaces → add a new Wireguard interface.
2. Add more than one peer for the interface.
3. Save and apply.
4. Navigate to the Peers tab of a freshly created interface.
5. Observe the issue: it is impossible to rearrange peers in LuCi without deleting them and creating in the right order.

## Actual behavior:
It is impossible to rearrange peers in LuCi without deleting them and creating in the right order.

## Expected behavior:
It is possible to rearrange peers right in LuCi.

closed 2024-10-25 by systemcrash

systemcrash, 2024-10-25:
Done 😄

this-username-has-been-taken, 2024-10-26:
Thank you very much! :)
It is now easier to manage peers. I hope others will also appreciate it as well!

commit f1b302d4 (2024-10-25): luci-proto-wireguard: make peer entries sortable

Closes #7354
commit 56294d01 (2024-10-25): luci-proto-wireguard: make peer entries sortable

Closes #7354

(cherry picked from commit f1b302d49bce91b0e25814af08c63681cf2744a5)