openwrt/luci #7355: luci-mod-dashboard: incorrect RSSI percentage

DragonBluep, 2024-10-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

[image]

### Actual behaviour

We are using this formula to calculate the RSSI value:
https://github.com/openwrt/luci/blob/0df7955d37e8db2fc18138186d8ff1113105c58b/modules/luci-mod-dashboard/htdocs/luci-static/resources/view/dashboard/include/30_wifi.js#L209

But some driver (e.g. mt76x02, rt2x00) doesn‘t support noise strength report. This will cause a wrong RSSI percentage.
```
phy0-ap0  ESSID: "PhantomXT-5G"
          Access Point: 
          Mode: Master  Channel: 36 (5.180 GHz)  HT Mode: VHT80
          Center Channel 1: 42 2: unknown
          Tx-Power: 24 dBm  Link Quality: 43/70
          Signal: -67 dBm  Noise: unknown
          Bit Rate: 523.2 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: 14C3:7662 14C3:7662 [MediaTek MT76x2E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy1-ap0  ESSID: "PhantomXT"
          Access Point: 
          Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: HT40
          Center Channel 1: 3 2: unknown
          Tx-Power: 30 dBm  Link Quality: 44/70
          Signal: -66 dBm  Noise: unknown
          Bit Rate: 111.1 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: embedded [MediaTek MT7620]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
```

### Expected behaviour

Do not show RSSI percentage or using some new formula to estimate the RSSI when noise is not supported.

### Steps to reproduce

No need.

### Additional Information

```shell
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r0-0a289bf3"
OPENWRT_BOARD="ramips/mt7620"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r0-0a289bf3"
OPENWRT_BUILD_DATE="1729816766"
```

### What browsers do you see the problem on?

Microsoft Edge

DragonBluep, 2024-10-26:
ping @systemcrash

DragonBluep, 2024-10-26:
And I suggest using -40 instead of -30 as the base offset. Usually -40dBm is already a very good signal. Maybe we also need to check if q > 100, then q = 100.

DragonBluep, 2024-10-26:
Another small issue about luci-mod-dashboard. The string "Transferred" can not be translated into the local language.
[image]

systemcrash, 2024-10-26:
So in the absence of a detected noise-floor, we default to an artificial noise floor of e.g. -90dBm. That should even out the calculations. 

-40 is a reasonable suggestion, although -30 is the established practical, which would avoid weird numbers like %110 if you're right next to the SSID and cross the antennae over.

commit 392b7e6b (2024-10-26): luci-mod-dashboard: fix for devices that do not report noise floor

Closes #7355
closed 2024-10-26 by systemcrash