openwrt/luci #7356: Feature request: Make Country Code a more important (or even mandatory) setting in LuCI

openwrtler, 2024-10-26:
### What would you like to see in luci?

Based on discussion [ [LuCI; Wi-Fi] Move Country Code from “Advanced Settings” to “General Setup”](https://forum.openwrt.org/t/luci-wi-fi-move-country-code-from-advanced-settings-to-general-setup/213227)

[Main points](https://forum.openwrt.org/t/luci-wi-fi-move-country-code-from-advanced-settings-to-general-setup/213227/14):
- Place Country code setting placement in main section instead of advanced section within LuCI, as country code seems mandatory for proper Wi-Fi functioning.
- Optionally make country code a global setting, as it probably is the same amongst all declared Wi-Fi networks for most users.
- Another idea is to make country code a required field in LuCI. If not set, display a warning for the user within LuCI.
- Current choice of Country code (no matter if "default driver", "World" or specific country or other default value) should be reflected in LuCI. In other words: I should not be able to select Wi-Fi channels (2.4/5 Ghz), that don't fit country code. Status quo seems to be [not restrictive enough](https://forum.openwrt.org/t/luci-wi-fi-move-country-code-from-advanced-settings-to-general-setup/213227/4).
- Some kind of first-time setup wizard including Country Code configuration would also be great, but probably more work to implement.

Depending on which features are implemented, changes of OpenWrt [quick start guide for Wi-Fi](https://openwrt.org/docs/guide-quick-start/walkthrough_wifi) to include a more understandable country code hint might lower barrier for router setups of new users.

systemcrash, 2024-10-26:
Sooo. Opinion time. 

I don't think that users should be limited - ever - in choices which won't degrade performance, at least. The whole point of openwrt is... "wireless freedom". The last thing the majority wants to mandates and restrictions again. The whole point of running openwrt is to avoid those, and free up your device to run a whole ecosystem. That said, users exercise their own judgement in their own domain what should be permissible.

What problem will mandating CC solve? Is this connected to some mongo hardware still going "you can't run 2.4GHz channel 14 unless you have JAPAN set." or so? Some users want it, some don't. 

Placement is discussable. I don't mind moving it - that's trivial, although muscle memory for some people will be cause for them to keep looking for CC in advanced :D

The wizard thing @stokito has a draught of I think.

hnyman, 2024-10-26:
If it is not mandatory in the underlying wifi driver, we should not make it mandatory in LuCI GUI.

But sure, it could be more prominently on the wifi config page front tab, as most/all users should set it.

dannil, 2024-10-27:
IMO we should probably move the country code to general instead of advanced settings. In addition to the quick start linked above, the guide to [Enabling a Wi-Fi access point on OpenWrt](https://openwrt.org/docs/guide-quick-start/basic_wifi) also advises users to set the country code. And not to mention that many users posting on the forums having issues with their WiFi are advised to check if the country code is set, such a basic setting to get a working AP should then not be in advanced settings.

systemcrash, 2024-10-28:
So is an acceptable solution move of CC from advanced tab to general?

hnyman, 2024-10-28:
Sounds good to me.

openwrtler, 2024-10-30:
Thanks for picking this up. I think it is a good first step usability-wise to move Country Code to General.

There is one pain point I'd still like to address (see also OP discussion post):
When creating a new Wi-Fi network via `Network` -> `Wireless` -> `radioX` -> `Add` *without* Country Code, network simply might not work for certain Wi-Fi channels and only show as `Disabled` in *Wireless Overview* of LuCI. Not sure, how underlying rules determine, if a channel is unsupported given no Country Code - all I can say is that Wi-Fi refused to work with 5GHz channels > 100. It worked with Channels < 100 and with auto-channel (which selected < 100). I found the hint `Disabled` to be a major source of confusion when debugging, up to point of wanting to return the router due to suspected radio controller defect.

Hence: What about adding an additional description to the network status in this case? Like:
> Disabled (unsupported or insufficient Country Code configuration)

@systemcrash Can you reopen this issue or does it make sense to open a separate one?

systemcrash, 2024-10-30:
State is state, no matter how you get there. Unless the underlying iwinfo or whichever daemon provides a specific return code, disabled means disabled. It doesn’t care whether you’re rich or poor… 

maybe you have a suggestion for some helpful text to accompany the setting? A call to action if you will.

openwrtler, 2024-10-31:
> Unless the underlying iwinfo or whichever daemon provides a specific return code, disabled means disabled.

I see. Are errors/return codes device-specific? In other words: Doesn't OpenWrt standardize common return codes/errors - in particular OP error, where a declared Wi-Fi network cannot be enabled due to channel restrictions in case of unconfigured Country Code? I indeed have hoped, this error can get caught separately.

> maybe you have a suggestion for some helpful text to accompany the setting?

Not sure what you are meaning here. I suggested 
> "Disabled (unsupported or insufficient Country Code configuration)"

text addition in previous post. Handling also depends on OpenWrt capabilities to distinguish Wi-Fi related errors. If we cannot address individual errors, my suggestion would be to add an info box/icon besides Wi-Fi status, which provides additional troubleshooting hints and which can be expanded to on hover/click (otherwise there probably is too much text going on in UI). 

A more general hint might be:
> Note: "Disabled" Wi-Fi network status can also be caused by an inproperly (or not at all) configured Country Code setting. Please set or change Country Code under "General" tab and retry enabling this Wi-Fi network.