openwrt/luci #7357: luci-app-opkg: PermissionError after upgrading from git-24.148.43905-2891ca4 to git-24.299.75545-b93fdef version

krisprols, 2024-10-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Actual behaviour

In System > Software menu. 
Click on the "Update Lists..." button.

I get the message "Unable to execute opkg update command: PermissionError: Access to command denied by ACL".

### Expected behaviour

LuCI should run "opkg update" successfully.

### Steps to reproduce

Upgrade luci-app-opkg from git-24.148.43905-2891ca4 to git-24.299.75545-b93fdef version.

Go to System > Software menu. 
Click on the "Update Lists..." button.

### Additional Information

```shell
NAME="OpenWrt"
VERSION="23.05.5"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 23.05.5"
VERSION_ID="23.05.5"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r24106-10cc5fcd00"
OPENWRT_BOARD="ath79/generic"
OPENWRT_ARCH="mips_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 23.05.5 r24106-10cc5fcd00"
```

### What browsers do you see the problem on?

Firefox

### Relevant log output

```shell
Unable to execute opkg update command: PermissionError: Access to command denied by ACL
```

systemcrash, 2024-10-26:
@Ansuel - it seems that update has extra space or parameters following the keyword 'update'. Maybe `/usr/libexec/opkg-call` actually takes `opkg update <pkgname>` after all?

systemcrash, 2024-10-26:
Tested on my VM. Upgrade went fine. 
```log
Upgrading luci-app-opkg on root from git-24.148.43905-2891ca4 to git-24.299.75545-b93fdef...
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/luci/luci-app-opkg_git-24.299.75545-b93fdef_all.ipk
Configuring luci-app-opkg.
```

systemcrash, 2024-10-26:
@krisprols  Did you change any of your json files or any custom edits before this?

systemcrash, 2024-10-26:
> Weirdly, I still see in the RPC calls e.g.:
> 
> ```shell
> /usr/libexec/opkg-call install --force-removal-of-dependent-packages luci-mod-network
> ```

Reboot fixed that :D

krisprols, 2024-10-26:
> @krisprols Did you change any of your json files or any custom edits before this?

Not at all. I could even reproduce the problem after I reseted my router.

krisprols, 2024-10-26:
> Tested on my VM. Upgrade went fine.
> 
> ```
> Upgrading luci-app-opkg on root from git-24.148.43905-2891ca4 to git-24.299.75545-b93fdef...
> Downloading https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/luci/luci-app-opkg_git-24.299.75545-b93fdef_all.ipk
> Configuring luci-app-opkg.
> ```

Yes the upgrade goes fine. That's not the problem. The problem is how this upgrade breaks the UI.

systemcrash, 2024-10-26:
I upgraded the package, rebooted, and could upgrade and install other packages. I repeated the test again just now. Same result. Works fine for me. Is your memory full or something?

Ansuel, 2024-10-26:
@systemcrash yes my bad i forgot to add the extra line from my testing env to the pr
the acl.d require to both handle single opkg update and opkg update *

systemcrash, 2024-10-26:
When I run opkg I get:
```
	update			Update list of available packages
	upgrade <pkgs>		Upgrade packages
```

update = No parameters 🤷

Ansuel, 2024-10-26:
What about update scenario? For an updated package? Is install called?

systemcrash, 2024-10-26:
> What about update scenario? For an updated package? Is install called?

That's surely

```
	upgrade <pkgs>		Upgrade packages
```

systemcrash, 2024-10-26:
Propose an edit for @krisprols to test first. See whether that fixes his problem?

krisprols, 2024-10-26:
> I upgraded the package, rebooted, and could upgrade and install other packages. I repeated the test again just now. Same result. Works fine for me. Is your memory full or something?

No, not a memory issue. 
[image]

krisprols, 2024-10-26:
FYI the problem only occurs for the  "Update Lists..." button. The "opkg update" command still works fine from command line. Installs and upgrades through GUI are fine as well. It looks like only the "opkg update" through the  "Update Lists..." button of the UI is impacted by this bug.

Ansuel, 2024-10-26:
Did you by chance cleared cache of the webpage? Tried incognito?

systemcrash, 2024-10-26:
Hmm. Is that json parameter something that sneaks in also?

Ansuel, 2024-10-26:
@systemcrash the parameter are generated by the .js page... If the old version is used then opkg-call update --force-removal-of... is still called...

systemcrash, 2024-10-26:
Yeah clear the browser cache. Delete all the cache for that router and reload. See whether that helps.

krisprols, 2024-10-26:
@Ansuel @systemcrash cleaning up the cache actually fixed the problem ! Thanks for the help. Sorry I haven't done that in the first place. I didn't bother trying as It looked like a server-side issue to me.

closed 2024-10-26 by krisprols

Ansuel, 2024-10-26:
@systemcrash well.... I think just to handle these sort of problem I would add the entry anyway just to catch these cache problem...
No idea what @jow- thinks about it.

(IN SHORT for jow:
we changed acl.d and we limit it to single opkg-call update but the .js might be cached and still forward additional args)

Silun, 2024-10-27:
I am getting the same issue: `Unable to execute opkg update command: PermissionError: Access to command denied by ACL`

I can confirm that the update is faulty. Not running out of cache or anything like that. Rebooting does not fix the issue. I cannot update my list any more via Luci after the update. I am on OpenWrt 23.05.5 r24106-10cc5fcd00 / LuCI openwrt-23.05 branch git-24.264.56413-c7a3562, router is AX3000T.

Deci8BelioS, 2024-10-27:
> I am getting the same issue: `Unable to execute opkg update command: PermissionError: Access to command denied by ACL`
>
> I can confirm that the update is faulty. Not running out of cache or anything [...]

The same thing happens to me, I get the error when I try to update the packages from the LuCi interface.

Silun, 2024-10-27:
I just downloaded and installed the generic older version still available in the repo [here](https://downloads.openwrt.org/releases/packages-23.05/aarch64_generic/luci/luci-app-opkg_git-24.148.43905-2891ca4_all.ipk) via SSH:

`wget https://downloads.openwrt.org/releases/packages-23.05/aarch64_generic/luci/luci-app-opkg_git-24.148.43905-2891ca4_all.ipk` to download the latest working version of the package, it might be different for your router, check carefully.

`opkg remove luci-app-opkg --force-depends` to force uninstall the current, broken version of the package.

`opkg install luci-app-opkg_git-24.148.43905-2891ca4_all.ipk` to install the working version from right before.

Et voila, it works again. If there's a better way, please do tell.

Deci8BelioS, 2024-10-27:
> I just downloaded and installed the generic older version still available in the repo [...]

Clearing the browser cache will fix it.

Silun, 2024-10-27:
Okay, yes, that worked. 🙈

goppinath, 2024-10-27:
I got the same [issue](https://forum.openwrt.org/t/unable-to-execute-opkg-update-command-permissionerror/214015), clearing the cache or accessing LuCi in the private window helps.

wlkmanist, 2024-10-28:
So, how it may be fixed without clearing the cache? Can't find an older version in packages.

rogovogor, 2024-10-28:
NOTE: You need to delete the cache, not the cookies

Ansuel, 2024-10-28:
@systemcrash Just to stop the ""flood"" I just reverted the commit for 23.xx and 22.xx. It's so minor it's not worth to partially revert and this won't happen on master or on the next release as we renamed the package (aka cache invalidated)

pawol, 2024-10-29:
I confirm that browser cache clearing helped to me with 3 openwrt devices