openwrt/luci #7358: luci: RPCError in "Network" tab

cocool97, 2024-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

```
RPCError

RPC call to uci/get failed with ubus code 9: Unspecified error
  at handleCallReply (http://192.168.1.1/luci-static/resources/rpc.js?v=git-24.264.56413-c7a3562:15:3)
```

### Actual behaviour

When navigating to "Network" tab in the luci web UI, I cannot access to "Interface" and "Routing" sub-items:
```
RPCError

RPC call to uci/get failed with ubus code 9: Unspecified error
  at handleCallReply (http://192.168.1.1/luci-static/resources/rpc.js?v=git-24.264.56413-c7a3562:15:3)
```

I did not notice any error on these pages before upgrading from a snapshot release (support for my router has just been added upstream)

I can provide more helpful information if needed, just tell me !

### Expected behaviour

Not raising an error and display the actual menu

### Steps to reproduce

go to Network menu -> Interfaces or Routing and see the error raising on web UI

### Additional Information

```shell
NAME="OpenWrt"
VERSION="23.05.5"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 23.05.5"
VERSION_ID="23.05.5"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r24106-10cc5fcd00"
OPENWRT_BOARD="mediatek/filogic"
OPENWRT_ARCH="aarch64_cortex-a53"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 23.05.5 r24106-10cc5fcd00"
```

### What browsers do you see the problem on?

Firefox

jow-, 2024-10-27:
Run a "uci show network >/dev/null" on the cli, see if a syntax error is reported. Repeat for wireless, system and dhcp.

cocool97, 2024-10-27:
Got following output:

```
root@OpenWrt:~# uci show network > /dev/null 
uci: Parse error (invalid character in name field) at line 36, byte 22
```

jow-, 2024-10-27:
So the problem is a syntax error in /etc/config/network, likely by editing the file manually outside of the gui.

closed 2024-10-27 by jow-

cocool97, 2024-10-27:
This was the point, thanks for the useful command !