openwrt/luci #7359: No VLAN for TP-Link TL-WR802N (may be issue in luci and/or core)

adrai, 2024-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

[image]

[image]

### Actual behaviour

Like suggested in [this original issue](https://github.com/openwrt/openwrt/issues/16792#issuecomment-2440193443), I'm creating also this issue here in the luci repository.

The TP-Link TL-WR802N seems to be supported, like listed here: https://openwrt.org/toh/tp-link/tl-wr802n_v4

But seems we can't make a VLAN setup. There is no check box to enable VLAN functionality and there are also wrong amount of ports, etc... see screenshots.

I would like to have different SSIDs, mapped to different VLANs.

### Expected behaviour

Expecting the Switch menu to have the VLAN functionality checkbox, so I can setup my SSIDs mapped with different VLANs

### Steps to reproduce

use a TP-Link TL-WR802N
go to the Switch menu, like in the screenshots

### Additional Information

```shell
NAME="OpenWrt"
VERSION="23.05.4"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 23.05.4"
VERSION_ID="23.05.4"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r24012-d8dd03c46f"
OPENWRT_BOARD="ramips/mt76x8"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 23.05.4 r24012-d8dd03c46f"
```

### What browsers do you see the problem on?

Firefox, Chrome, Safari, Microsoft Edge

LGA1150, 2024-11-11:
AFAIK it's a unknown issue. luci assumes _VLAN index_ equals _VLAN ID_. In some old hardware there are limited indices, and in your case, 0~15. So you have to assign a different VLAN ID to a VLAN index using an `option vid`.

https://openwrt.org/docs/guide-user/network/vlan/switch#vlanswitch_config

adrai, 2024-11-11:
But why is there no VLAN checkbox in the UI?
And why is there that warning message with unknown topology?

adrai, 2024-11-11:
```sh
BusyBox v1.36.1 (2024-07-15 22:14:18 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.4, r24012-d8dd03c46f
 -----------------------------------------------------
root@WLAN-Access-Point-OpenWrt:~# swconfig list
Found: switch0 - rt305x
```

LGA1150, 2024-11-11:
It looks like the switch device name does not match the config in `/etc/config/network`. The config should be `switch0` instead of `eth0`.
```
config switch
	option name 'switch0'
	option enable_vlan '1'
```

adrai, 2024-11-11:
just renamed it and then `/etc/init.d/network restart` and now the device is not reachable anymore

LGA1150, 2024-11-11:
It worked - but you didn't configure any VLANs so it drops all packets..

adrai, 2024-11-11:
the port should be configured as untagged... so with vlan enabled, it should work, right?
Or can you share a network config file that you think should work?
I like the port not to be tagged by default, but only the packets should be tagged, based on the ssid...

LGA1150, 2024-11-11:
You may create a default VLAN 1 for further configuration.

```
config switch
	option name 'switch0'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '0 1 2 3 4 6'
```

LGA1150, 2024-11-11:
Wait a minute.. WR802N has a single ethernet port, right? Then you don't even need switch VLANs. Just create an interface and set the device to `eth0.yourVlanID`

adrai, 2024-11-11:
Yes, there is only 1 eth port.
But it shows 7 in the UI.

Can you share the complete config, please?

LGA1150, 2024-11-11:
```
config device
	option name 'br-guests'
	option type 'bridge'
	list ports 'yourVlanID'

config interface 'guests'
	option device 'br-guests'
	option proto 'static'
	option ipaddr 'yourIp'
	option netmask 'yourNetmask'
```

adrai, 2024-11-11:
Seems to work... thank you 🙏 

Now just the little issue regarding the switch menu with the 7 ports instead of 1.

Can this also be "fixed" via config?

LGA1150, 2024-11-11:
The switch menu is useless now. You may consider removing it.
`opkg remove swconfig`

adrai, 2024-11-11:
perfect, thank you

closed 2024-11-11 by adrai