openwrt/luci #7362: DHCP and DNS Dnsmasq is a lightweight DHCP server and DNS forwarder.

Dexterity128, 2024-10-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

no

### Actual behaviour

Maximum allowed number of concurrent DNS queries is too low at 150 it caused router to be disconnected from the internet

### Expected behaviour

Maximum allowed number of concurrent DNS queries at default should be 300

### Steps to reproduce

Network- then DHCP and DNS

### Additional Information

```shell
Status
System
Hostname	OpenWrt
Model	TP-Link Archer C7 v5
Architecture	Qualcomm Atheros QCA956X ver 1 rev 0
Target Platform	ath79/generic
Firmware Version	OpenWrt 23.05.5 r24106-10cc5fcd00 / LuCI openwrt-23.05 branch git-24.264.56413-c7a3562
Kernel Version	5.15.167
Local Time	2024-10-29 15:34:02
Uptime	0h 38m 29s
Load Average	0.06, 0.02, 0.00
```

### What browsers do you see the problem on?

Microsoft Edge

systemcrash, 2024-10-29:
This may get changed in the openwrt repo when 8/64 MB targets get dropped after 23. But in the meantime it's fully within your control in the GUI. Set it higher. DHCP -> Limits.

closed 2024-10-29 by systemcrash

geldot, 2024-11-10:
> ### Actual behaviour
> Maximum allowed number of concurrent DNS queries is too low at 150 it caused router to be disconnected from the internet

The "Maximum number of concurrent DNS queries reached (max: 150)" error is almost certainly a symptom, not the cause, of an internet disconnection. When dnsmasq is unable to forward requests due to upstream DNS servers becoming unreachable, it will inevitably issue this message when the number of pending queries is reached. Hence, the message itself is confusing but harmless and raising the setting from 150 to 300 won't help.

systemcrash, 2024-11-10:
...and nothing to do with the luci repo.