openwrt/luci #7372: Feature request: temperature sensors in luci

damnms, 2024-10-31:
### What would you like to see in luci?

I just stumbled upon https://github.com/gSpotx2f/packages-openwrt/raw/master/current/luci-app-temp-status_0.4.1-r1_all.ipk and wondering if such feature could be implemented in luci. i am a bit too afraid to install an ipk from someone i dont know or whats not within the official openwrt distribution. It would be great to see cpu/nvme/ssd/sfp etc. temperatures.

dannil, 2024-11-01:
Can you check if `luci-app-statistics` fulfill your needs? It should have temperature sensors functionality. You would need to install `collectd-mod-thermal` as well, see below for the docs.

https://openwrt.org/docs/guide-user/luci/luci_app_statistics

damnms, 2024-11-01:
looks good! i was not aware that such plugin exists. Unfortunately, its missing some data, e.g. the sfp+ port

hnyman, 2024-11-02:
> its missing some data, e.g. the sfp+ port

For getting data from some temperature (and other) sensors, you need to use the generic sensors plugin instead.
Install both "lm-sensors" and "collectd-mod-sensors"

hnyman, 2024-11-02:
I close the feature request, as there are already both the CPU temp monitoring and the generic sensors monitoring available in the LuCI statistics.

closed 2024-11-02 by hnyman