openwrt/luci #7379: Feature request: ability to operate and control GPIO from luci.

professor-jonny, 2024-11-05:
### What would you like to see in luci?

I have a modem with an external antenna it uses GPIO to switch between the internal and external antenna I'd like to have a menu item in luci (like the LED menu) to be able to assign triggers or virtual buttons or switches to be able to control the antenna or any general purpose IO that is exported to userspace.

professor-jonny, 2024-11-05:
Currently the device I'm looking at is pending new target.
https://github.com/openwrt/openwrt/pull/16329

```
		ext_antenna_en {
			gpio-export,name = "ext_antenna_en";
			gpio-export,output = <0>;
			gpios = <&tlmm 56 GPIO_ACTIVE_HIGH>; /* select external sma antennas for wwan output */
```

I'd imagine there is a few devices that have the same issue now as there are a few devices with built in modems with external antennas.
 
 https://openwrt.org/tag/lte?do=showtag&tag=lte
 
Almost all the devices have GPIO to control the power to the LTE modem to enable it and some have a reset GPIO.

systemcrash, 2024-11-05:
The GUI can have anything that the openwrt system itself is able to handle: so if there are scripts or functionality in e.g. modemmanager, those can be triggered from its GUI.

professor-jonny, 2024-11-05:
There are no scripts for the gpio to be controlled as such, in the case of this device is just exported to userspace I guess that is the case with most.

It seems the modem power is permanently enabled from every device with a default on issue from userspace or the dts.

If one wishes to enable or disable the external antenna currently one must type echo commands into a terminal pointing at the appropriate sysfs property to enable or disable the GPIO or alter the init/ boot scripts.

It would be nice to include a new menu item in luci with the ability to directly control gpio that is tied to features like modem modem power, reset modem, external antenna, there is possibly ither things enable disable poe power to lan/wan switch ports cooling fan's etc....

It would not be a silly idea to include all IO, it would help somewhat in development, as one could trigger inputs and outputs visually watch and control it from luci, but hat is a different use case and would not be needed for most users (dev option).

systemcrash, 2024-11-05:
Welp, all guard rails are off, in that case. Sending the wrong signal to a GPIO can, at worst, reset your device. I have difficulty to see specific controls, but rather a generalised "send this" to "this GPIO".

professor-jonny, 2024-11-06:
In this case I'm only really interested in the exported GPIO. and they are all related to the modem for my device.
Ideally a button form luci to offer pulse and toggle options of these GPIO to switch my modem to use an external antenna or depower the modem without depowering the complete device.

But I can understand adjusting the NAND address lines or something is a bad idea and I'm not proposing that, but in saying that ability to view and possibly control GPIO from an intitramfs for dev purposes could be handy.

feckert, 2024-11-06:
@professor-jonny I see it the same way as @systemcrash. If we bring this into the LuCI, then a lot of nonsense can be done with it. So I don't think we should to this. The problem is also that the current implementation `/sys/class/gpio` is marked as [obsolet](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/Documentation/ABI/obsolete/sysfs-gpio?h=v5.16&id=fe95046e960b4b76e73dc1486955d93f47276134) in the kernel.

Also, this is all device dependent and there is no generic implementation for this. If we want to support this in a generic form then we should write a script or a program that has a generic API and then we can add devices to them to switch a modem on/off or switch the antenna for a modem.

A fast solution for you would be to write a script here to abstract the implementation. This can be called then via [luci-app-command](https://github.com/openwrt/luci/tree/master/applications/luci-app-commands).

closed 2024-11-06 by feckert