openwrt/luci #7388: Feature request: better Wi-Fi DFS state display

k-korn, 2024-11-13:
### What would you like to see in luci?

Currently, if the user selects one of 5GHz DFS channels (52-144 in US, for example), DFS behavior is completely opaque to the user:
- If the hardware supports DFS scans, the access point will come online only after the scan is done (~60 seconds).
- If the hardware does NOT support DFS, the access point will never come online and the interface will be left inactive indefinitely - with no indication in the UI.
- Yes, the issue can be dealt with by looking at hostapd logs - but one must be aware of this behavior to successfully fix the issue and move the AP to different channel.

My suggestion would be to improve visibility of DFS state in luci:
- At minimum, indicate DFS channels in the list, for example  "132 (5666 MHz, DFS)":
[image]

- If possible, provide current state of DFS in some notable UI place (Wi-Fi Overview page?).
For the following log, the state should change "DFS Enabled -> DFS Scanning -> DFS OK"
```
Wed Nov 13 15:08:05 2024 daemon.notice hostapd: wlan5: interface state HT_SCAN->DFS
...
Wed Nov 13 15:09:11 2024 daemon.notice hostapd: wlan5: interface state DFS->ENABLED
Wed Nov 13 15:09:11 2024 daemon.notice hostapd: wlan5: AP-ENABLED

```
- If DFS scan has been initiated but never finished (if the hardware does not support DFS), the interface should indicate something like "DFS FAILED" in red, suggesting channel change.

I am, unfortunately, too far from UI / Luci development to actually implement this - but at least, here is a suggestion.

systemcrash, 2024-11-13:
@Ansuel didn't your earlier fix which I merged for outdoor channels open the possibility to this? #5695

@dannil or @knarrff is this something interesting for you?

Ansuel, 2024-11-13:
@systemcrash well yes the info are all there... about DFS state... it might require some additional support from the mac80211 side and rpcd but I think it's doable and it's actually a not so bad feature

dannil, 2024-11-15:
As long as we can make the iwinfo rpcd call to return the DFS channels in an uniform way independent of underlying driver (ath11k, mt76 etc.) this shouldn't be hard to fix, I like the idea as well.

rmandrad, 2024-11-21:
I create a [pull ](https://github.com/openwrt/iwinfo/pull/18)on iwinfo to display the DFS channels

systemcrash, 2024-11-21:
badass. Thanks @rmandrad

systemcrash, 2024-12-30:
So apparently RADAR_DETECTION based channels are now easily detectable thanks to ucode additions. 

https://github.com/openwrt/openwrt/blob/874e0accae5121534259cc8b1bd3107bd104dd24/package/network/config/wifi-scripts/files-ucode/usr/share/ucode/iwinfo.uc#L344

nekohayo, 2026-09-06:
It would be really nice if the DFS ranges were also represented in `/cgi-bin/luci/admin/status/channel_analysis`, using some sort of subtle [hatching](https://en.wikipedia.org/wiki/Hatching) pattern rectangle region or something like that. Same idea as in https://github.com/yeager/wifi-analyzer/issues/9.