openwrt/luci #7389: opkg update fails with the following error below:

James62370, 2024-11-13:
Executing package manager
Downloading https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/packages/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/packages/Packages.gz

Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz

Downloading https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/kmods/6.6.57-1-a15c2efc06adccbc75cbd04e08bd4017/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_kmods
Downloading https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/kmods/6.6.57-1-a15c2efc06adccbc75cbd04e08bd4017/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.gz

Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz

Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz

Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz
Errors
Collected errors:
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz, wget returned 8.

-------------------------------------------------------
I tried going back a few versions and I get the same error on my Access Point and Mesh Nodes.  However, everything seems to still work correctly.

Any insight or help in resolving this would be appreciated.

Thank you in advance!

hnyman, 2024-11-13:
The cure is to update to the current up-to-date version of OpenWrt testing builds, that do not have opkg any more, but apk instead.
(there is no opkg support any more in main/master snapshots)

Nothing to do with LuCI GUI in any case, so closing this.

closed 2024-11-13 by hnyman

James62370, 2024-11-13:
Appreciate the reply.  However, I don't quite understand what you mean?  I use to upgrade using sysupgrade within luci.  I'm on pretty recent snapshots.  Could you please elaborate on where I can find the "current up-to-date version of OpenWrt testing builds"?

hnyman, 2024-11-13:
opkg was switched to apk 3 days ago.

> where I can find the "current up-to-date version of OpenWrt testing builds"?

https://downloads.openwrt.org/
Specifically https://downloads.openwrt.org/snapshots/targets/

James62370, 2024-11-13:
Is there a simple way to replace opkg with apk?  From some of the reddit posts that I've read, it this something not completely implemented as of yet?  Source: https://www.reddit.com/r/openwrt/comments/1gokyrg/the_apk_package_manager_is_now_the_default/

hnyman, 2024-11-13:
The simple way is to sysupgrade to a new build.
Nothing else is really supported.

James62370, 2024-11-13:
I appreciate all the replies.  Just to clarify, attended sysupgrade through the luci interface won't resolve this issue?  I'm testing with a spare Dynalink DL-WRX36 Router (qualcommax/ipq807x) as mesh node.  The latest SNAPSHOT r28062-be3e7a84bd doesn't appear to resolve the opkg to apk issue.

Is the only upgrade path starting from scratch?  Then do an apk update (previously opkg update) and install each package manually again and restoring my configuration?

hnyman, 2024-11-14:
normal sysupgrade works.

Attendedsysupgrade & auc & owut & whatever...  tools for quickly cooking up tailored firmware are still being adapted to apk.

> only upgrade path starting from scratch? Then do an apk update (previously opkg update) and install each package manually again and restoring my configuration?

That should work.

dangerweenie, 2025-01-03:
I updated to the new version that has both opkg and apk and when I try to search for package in Luci in either package manager it throws errors (like the repositories are wrong).  Can't seem to figure it out.

They're both using /etc/apk/repositories.d/distfeeds.list which is populated with the path that has my router name in it (i.e. Qualcomm/ip86 or something to that effect.

hnyman, 2025-01-03:
> updated to the new version that has both opkg and apk 

None of OpenWrt versions should have both APK and opkg enabled.
And there is only one packagemanager LuCI app.

I suggest that you download a normal main/master snapshot image and sysupgrade	 to it in order to get into the new baseline. Then add new packages to the system.