openwrt/luci #7391: Feature request: Actual information on help

gcb, 2024-11-14:
### What would you like to see in luci?

There are some hints on fields which are redundant, some examples:

[Metric: Specifies the route metric to use](https://github.com/openwrt/luci/blob/d6506880edb0cfbd8e94790911a353a57b23f5c4/modules/luci-mod-network/htdocs/luci-static/resources/view/network/routes.js#L81)

[MTU: Defines a specific MTU for this route](https://github.com/openwrt/luci/blob/d6506880edb0cfbd8e94790911a353a57b23f5c4/modules/luci-mod-network/htdocs/luci-static/resources/view/network/routes.js#L88)

Those are the minority, but maybe we can add more value to them. For example, i always forget if route metrics are importance level (lower means later) or order (lower means sooner)... but the hint doesn't mention either way. same for MTU.

systemcrash, 2024-11-14:
I believe the term you're looking for is ordinal. 1st, 2nd, 3rd. i.e. order (so lowest wins).

What would you propose for MTU? Something a beginner would benefit from.

gcb, 2024-11-14:
exactly! i put this up as a reminder to write something in a PR later... My plan was to look at whats already written elsewhere.  But the first place I looked https://openwrt.org/docs/guide-user/network/network_configuration suffers from a similar problem :)

For MTU i would probably suggest something on the lines of "override Maximum Transmission Unit size in bytes(?). Not recommended".

gcb, 2024-11-14:
for metric best i found on openwrt.org is https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3?s[]=route&s[]=metric

> The default (primary) WAN interface should have the lowest metric. Values are not important, but should always be unique.

just removing the wan mentions there should be good.

systemcrash, 2024-11-14:
Sometimes developers write incomprehensible stuff inaccessible to other users. Sure it's logical what it does, but what is the consequence, y'know?

Closed by a9e1685f6419432c5ce682895d44a2d14eb16fcc

closed 2024-11-14 by systemcrash