openwrt/luci #7395: Luci-Theme: LuCI Master 24.319.72217~9fe1e34 is not providing theme selections in the theme dropdown

Daprogra2, 2024-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

[image]
[image]
[image]

LuCI Master 24.319.72217~9fe1e34 is not providing theme selections in the theme dropdown. This is preventing the any settings changes in the System Tab from being saved. 

### Actual behaviour

1. Opened System menu in LUCI
2. Changed router name <General Settings - Hostname>
3. Click **Save and Apply**
4. Received error - **_An error occurred while saving the form: Option "Design" contains an invalid input value. Expecting: non-empty value_**

### Expected behaviour

1. Opened System menu in LUCI
2. Changed router name _**<General Settings - Hostname>**_
3. Click _**Save and Apply**_
4. Name Change Is Successful
  

### Steps to reproduce

1. Opened System menu in LUCI
2. Changed router name <General Settings - Hostname>
3. Click **Save and Apply**

### Additional Information

```shell
root@OpenWrt:~# cat /etc/os-release
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28085-6720c4ccba"
OPENWRT_BOARD="mediatek/filogic"
OPENWRT_ARCH="aarch64_cortex-a53"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r28085-6720c4ccba"
OPENWRT_BUILD_DATE="1731585818"

LuCI Master 24.319.72217~9fe1e34
```

### What browsers do you see the problem on?

Firefox

### Relevant log output

```shell
<select id="widget.cbid.luci.cfg01e48a._mediaurlbase" class="cbi-input-select cbi-input-invalid" data-tooltip="Expecting: non-empty value" data-tooltip-style="error"></select>

Uncaught (in promise) TypeError: Option "Design" contains an invalid input value. Expecting: non-empty value
    parse http://192.168.60.1/luci-static/resources/form.js?v=24.319.72217~9fe1e34:137
    parse http://192.168.60.1/luci-static/resources/form.js?v=24.319.72217~9fe1e34:71
    parse http://192.168.60.1/luci-static/resources/form.js?v=24.319.72217~9fe1e34:47
    save http://192.168.60.1/luci-static/resources/form.js?v=24.319.72217~9fe1e34:47
    callClassMethod http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:137
    handleSave http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:147
    handleSave http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:147
    handleSaveApply http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:147
    createHandlerFn http://192.168.60.1/luci-static/resources/ui.js?v=24.319.72217~9fe1e34:360
    handleClick http://192.168.60.1/luci-static/resources/ui.js?v=24.319.72217~9fe1e34:160
    bind http://192.168.60.1/luci-static/resources/ui.js?v=24.319.72217~9fe1e34:58
    render http://192.168.60.1/luci-static/resources/ui.js?v=24.319.72217~9fe1e34:47
    render http://192.168.60.1/luci-static/resources/ui.js?v=24.319.72217~9fe1e34:157
    addFooter http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:149
    __init__ http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:147
    promise callback*__init__ http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:147
    super http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:22
    ClassConstructor http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:12
    compileClass http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:182
    promise callback*compileClass http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:178
    promise callback*require http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:184
    instantiateView http://192.168.60.1/luci-static/resources/ui.js?v=24.319.72217~9fe1e34:360
    <anonymous> http://192.168.60.1/cgi-bin/luci/admin/system/system:51
    promise callback* http://192.168.60.1/cgi-bin/luci/admin/system/system:50
luci.js:137:348
Uncaught (in promise) TypeError: Option "Design" contains an invalid input value. Expecting: non-empty value
    parse http://192.168.60.1/luci-static/resources/form.js?v=24.319.72217~9fe1e34:137
    parse http://192.168.60.1/luci-static/resources/form.js?v=24.319.72217~9fe1e34:71
    parse http://192.168.60.1/luci-static/resources/form.js?v=24.319.72217~9fe1e34:47
    save http://192.168.60.1/luci-static/resources/form.js?v=24.319.72217~9fe1e34:47
    callClassMethod http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:137
    handleSave http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:147
    handleSave http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:147
    handleSaveApply http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:147
    createHandlerFn http://192.168.60.1/luci-static/resources/ui.js?v=24.319.72217~9fe1e34:360
    handleClick http://192.168.60.1/luci-static/resources/ui.js?v=24.319.72217~9fe1e34:160
    bind http://192.168.60.1/luci-static/resources/ui.js?v=24.319.72217~9fe1e34:58
    render http://192.168.60.1/luci-static/resources/ui.js?v=24.319.72217~9fe1e34:47
    render http://192.168.60.1/luci-static/resources/ui.js?v=24.319.72217~9fe1e34:157
    addFooter http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:149
    __init__ http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:147
    promise callback*__init__ http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:147
    super http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:22
    ClassConstructor http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:12
    compileClass http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:182
    promise callback*compileClass http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:178
    promise callback*require http://192.168.60.1/luci-static/resources/luci.js?v=24.319.72217~9fe1e34:184
    instantiateView http://192.168.60.1/luci-static/resources/ui.js?v=24.319.72217~9fe1e34:360
    <anonymous> http://192.168.60.1/cgi-bin/luci/admin/system/system:51
    promise callback* http://192.168.60.1/cgi-bin/luci/admin/system/system:50
luci.js:137:348
```

hnyman, 2024-11-16:
You may have found something.

I see the normal selection between the Bootstrap variants.
But if I try to install openwrt-2020 theme, there is an error and the package does not appear into the dropdown selection list.  Neither is there any trace in /etc/config/luci, although there are openwrt-2020 theme files in /overlay/upper.

The error given is:

```
Executing package manager

apk add luci-theme-openwrt-2020

(1/1) Installing luci-theme-openwrt-2020 (24.320.57985~1a4b6ad)
Executing luci-theme-openwrt-2020-24.320.57985~1a4b6ad.post-install
1 error; 71 MiB in 309 packages

The apk install command failed with code 1.
```

Looks like it fails to run the post-install actions, or something like that.
The error message does not appear more clear even from ssh console with --verbose.

systemcrash, 2024-11-16:
Looks like fallout from apk switchover?

Daprogra2, 2024-11-16:
I was able to install a  couple of themes via LUCI but they did not show in the dropdown

hnyman, 2024-11-17:
I tested a bit with r28102-65964c42f8 and noticed that 
 * the script setting the config values to /etc/config/luci was placed into /etc/uci-defaults directory,   (https://github.com/openwrt/luci/blob/master/themes/luci-theme-openwrt-2020/root/etc/uci-defaults/30_luci-theme-openwrt-2020 )
 * but **it was not being run** along the theme installation.  opkg or apk should run it at installation.
 
When I then booted the router, the theme got changed and both themes were visible in the menu.

This is probably something that @aparcar and @Ansuel are interested, as it suggests that the apk package installation process does not currently do quite the same steps as opkg did.

hnyman, 2024-11-17:
The root cause may be 
* some kind of incompatibility between theme packages and https://github.com/openwrt/openwrt/blob/main/package/base-files/files/lib/functions.sh   , or 
* apk not running the default-postinst in /lib/functions.sh at all. (I tried adding debug statements there, but saw no output. Looks like that default-postinst is not reached)  
cc @aparcar   Any ideas?

robimarko, 2024-11-21:
This should get fixed by https://github.com/openwrt/openwrt/pull/17023

Testing is much appreciated

hnyman, 2024-11-21:
@robimarko 

I am not yet sure if this is solved at r28144-2784f78c6c, self-built for MT6000.

I tested installing luci-theme-material, and the config lines did not appear into /etc/config/luci.

The theme has been added to overlay on flash:
```
root@router6000:~# ls -l /overlay/upper/www/luci-static/
drwxr-xr-x    3 root     root          3488 Nov 21 20:40 material
drwxr-xr-x    2 root     root          3488 Nov 21 20:40 resources
```
```
root@router6000:~# apk list *theme*
luci-theme-bootstrap-24.284.01692~c68d5c8 noarch {feeds/luci/themes/luci-theme-bootstrap} (Apache-2.0) [installed]
luci-theme-bootstrap-24.322.80622~a403707 all {feeds/luci/themes/luci-theme-bootstrap} (Apache-2.0) [upgradable from: luci-theme-bootstrap-24.284.01692~c68d5c8]
luci-theme-material-24.322.80622~a403707 all {feeds/luci/themes/luci-theme-material} (Apache-2.0) [installed]
luci-theme-openwrt-24.322.80622~a403707 all {feeds/luci/themes/luci-theme-openwrt} (Apache-2.0)
luci-theme-openwrt-2020-24.322.80622~a403707 all {feeds/luci/themes/luci-theme-openwrt-2020} (Apache-2.0)
```
And the not-yet-executed uci-defaults script is visible:

```
root@router6000:~# cat /etc/uci-defaults/30_luci-theme-material
#!/bin/sh

if [ "$PKG_UPGRADE" != 1 ]; then
        uci get luci.themes.Material >/dev/null 2>&1 || \
        uci batch <<-EOF
                set luci.themes.Material=/luci-static/material
                set luci.main.mediaurlbase=/luci-static/material
                commit luci
        EOF
fi

exit 0
```

hnyman, 2024-11-21:
I likely figured one possible explanation:
LuCI themes do not define postinst function, but instead they directly add /etc/uci-defaults file, which was automatically ran by opkg installation, but apparently not by apk.

Likely the themes should be converted to declare postinst function in Makefile, like they actually do for postrm...

systemcrash, 2024-11-21:
Do you think this is unique to themes?

robimarko, 2024-11-21:
I explicitly tested with luci-theme-material and uci-defaults script was for sure executed as UCI config was updated and newly installed theme was default, prior to these changes that would not happen and script would still be there.

But do note that you have to build the packages yourself as well as those that are currently in feeds are broken and it will take some time for SDK-s to get regenerated and then packages rebuilt.

robimarko, 2024-11-21:
For example with the 2020 theme, you can clearly see it works
```
root@Dynalink:/tmp# cat /etc/config/luci

config core 'main'
	option lang 'auto'
	option mediaurlbase '/luci-static/material'
	option resourcebase '/luci-static/resources'
	option ubuspath '/ubus/'

config extern 'flash_keep'
	option uci '/etc/config/'
	option dropbear '/etc/dropbear/'
	option openvpn '/etc/openvpn/'
	option passwd '/etc/passwd'
	option opkg '/etc/opkg.conf'
	option firewall '/etc/firewall.user'
	option uploads '/lib/uci/upload/'

config internal 'languages'

config internal 'sauth'
	option sessionpath '/tmp/luci-sessions'
	option sessiontime '3600'

config internal 'ccache'
	option enable '1'

config internal 'themes'
	option Bootstrap '/luci-static/bootstrap'
	option BootstrapDark '/luci-static/bootstrap-dark'
	option BootstrapLight '/luci-static/bootstrap-light'

config internal 'apply'
	option rollback '90'
	option holdoff '4'
	option timeout '5'
	option display '1.5'

config internal 'diag'
	option dns 'openwrt.org'
	option ping 'openwrt.org'
	option route 'openwrt.org'

root@Dynalink:/tmp# wget http://192.168.3.32:8000/luci-theme-openwrt-2020-24.326.01848~53e36e3.apk
Downloading 'http://192.168.3.32:8000/luci-theme-openwrt-2020-24.326.01848~53e36e3.apk'
Connecting to 192.168.3.32:8000
Writing to 'luci-theme-openwrt-2020-24.326.01848~53e36e3.apk'
luci-theme-openwrt-2 100% |*******************************| 45812   0:00:00 ETA
Download completed (45812 bytes)
root@Dynalink:/tmp# apk add luci-theme-openwrt-2020-24.326.01848\~53e36e3.apk 
(1/1) Installing luci-theme-openwrt-2020 (24.326.01848~53e36e3)
Executing luci-theme-openwrt-2020-24.326.01848~53e36e3.post-install
OK: 31 MiB in 172 packages
root@Dynalink:/tmp# ls /etc/uci-defaults/
root@Dynalink:/tmp# cat /etc/config/luci

config core 'main'
	option lang 'auto'
	option mediaurlbase '/luci-static/openwrt2020'
	option resourcebase '/luci-static/resources'
	option ubuspath '/ubus/'

config extern 'flash_keep'
	option uci '/etc/config/'
	option dropbear '/etc/dropbear/'
	option openvpn '/etc/openvpn/'
	option passwd '/etc/passwd'
	option opkg '/etc/opkg.conf'
	option firewall '/etc/firewall.user'
	option uploads '/lib/uci/upload/'

config internal 'languages'

config internal 'sauth'
	option sessionpath '/tmp/luci-sessions'
	option sessiontime '3600'

config internal 'ccache'
	option enable '1'

config internal 'themes'
	option Bootstrap '/luci-static/bootstrap'
	option BootstrapDark '/luci-static/bootstrap-dark'
	option BootstrapLight '/luci-static/bootstrap-light'
	option OpenWrt2020 '/luci-static/openwrt2020'

config internal 'apply'
	option rollback '90'
	option holdoff '4'
	option timeout '5'
	option display '1.5'

config internal 'diag'
	option dns 'openwrt.org'
	option ping 'openwrt.org'
	option route 'openwrt.org'
```

hnyman, 2024-11-21:
> note that you have to build the packages yourself as well as those that are currently in feeds are broken and it will take some time for SDK-s to get regenerated and then packages rebuilt.

Thanks for explaining. 
I forgot that the .apk package itself needs fix/recompilation.

I tested again with locally built .apk, and it works ok:
```
root@router4:~# apk add /tmp/luci-theme-material-24.191.51591\~5119fa2.apk
(1/1) Installing luci-theme-material (24.191.51591~5119fa2)
Executing luci-theme-material-24.191.51591~5119fa2.post-install
OK: 64 MiB in 318 packages
```
```
config internal 'themes'
        option Bootstrap '/luci-static/bootstrap'
        option OpenWrt2020 '/luci-static/openwrt2020'
        option BootstrapDark '/luci-static/bootstrap-dark'
        option BootstrapLight '/luci-static/bootstrap-light'
        option Material '/luci-static/material'
```

thanks for hint.

I will mark this solved, although it may take 2-3 days before the fixed packages are publicly available from the download site for all architectures.

closed 2024-11-21 by hnyman