openwrt/luci #7397: Feature request: <allow to request specific ipv6 prefix>

try496, 2024-11-16:
### What would you like to see in luci?

odhcp6c has merged the commit that supports
this feature. If the ISP supports this feature, it
would allow obtaining a specific IPv6 prefix. I
hope that luci will addd aan option for it.

try496, 2024-11-19:
After testing, it has been confirmed that by requesting a specified length of an IPv6 prefix and filling in the corresponding address, one can obtain the desired IP address or prefix

try496, 2024-11-19:
Perhaps the text could be updated to better illustrate this point.

systemcrash, 2024-11-19:
That setting has only recently been changed in odhcp6c.

Previously.
```
Usage: odhcp6c [options] <interface>

Feature options:
...
	-P <length>	Request IPv6-Prefix (0 = auto)
```

After this commit:
https://github.com/openwrt/odhcp6c/commit/b6ae9ffaeb0e18e9fa3d5be62faa8d3c9f340a58

```
Usage: odhcp6c [options] <interface>

Feature options:
...
	-P <[pfx/]len>	Request IPv6-Prefix (0 = auto)
```

try496, 2024-11-19:
https://github.com/openwrt/openwrt/commit/b0312c10818f36bc6e61a518bb528d2809562796   The latest commit has been merged into the snapshot.

try496, 2024-11-19:
Perhaps we can wait until the 24.10 branch is merged to update the description.

systemcrash, 2024-11-19:
That's in main, not in snapshot. 

> [openwrt/openwrt@b0312c1](https://github.com/openwrt/openwrt/commit/b0312c10818f36bc6e61a518bb528d2809562796) The latest commit has been merged into the snapshot.

Go ahead and cherry-pick it to snapshot (24) also.

If it's in main, we can at least modify this string in main here.

ynezz, 2024-11-19:
> to what?

```diff
diff --git a/protocols/luci-proto-ipv6/htdocs/luci-static/resources/protocol/dhcpv6.js b/protocols/luci-proto-ipv6/htdocs/luci-static/resources/protocol/dhcpv6.js
index ff242bb12c96..e00c8c762940 100644
--- a/protocols/luci-proto-ipv6/htdocs/luci-static/resources/protocol/dhcpv6.js
+++ b/protocols/luci-proto-ipv6/htdocs/luci-static/resources/protocol/dhcpv6.js
@@ -20,9 +20,10 @@ return network.registerProtocol('dhcpv6', {
 		o.value('none', _('disabled', 'DHCPv6 address request mode'));
 		o.default = 'try';
 
-		o = s.taboption('general', form.Value, 'reqprefix', _('Request IPv6-prefix of length'));
+		o = s.taboption('general', form.Value, 'reqprefix', _('Request IPv6-prefix'),
+						_('Number for hinted prefix length (0-64) or specific prefix/length (e.g. 2001:db8::/56)'));
 		o.value('auto', _('Automatic'));
-		o.value('no', _('disabled'));
+		o.value('no', _('Only an address (no subnet/routing)'));
 		o.value('48');
 		o.value('52');
 		o.value('56');
```

systemcrash, 2024-11-19:
Agreed that we should use a descriptor, and not title.

systemcrash, 2024-11-19:
```diff
diff --git a/protocols/luci-proto-ipv6/htdocs/luci-static/resources/protocol/dhcpv6.js b/protocols/luci-proto-ipv6/htdocs/luci-static/resources/protocol/dhcpv6.js
index ff242bb12c96..e00c8c762940 100644
--- a/protocols/luci-proto-ipv6/htdocs/luci-static/resources/protocol/dhcpv6.js
+++ b/protocols/luci-proto-ipv6/htdocs/luci-static/resources/protocol/dhcpv6.js
@@ -20,9 +20,10 @@ return network.registerProtocol('dhcpv6', {
 		o.value('none', _('disabled', 'DHCPv6 address request mode'));
 		o.default = 'try';
 
-		o = s.taboption('general', form.Value, 'reqprefix', _('Request IPv6-prefix of length'));
+		o = s.taboption('general', form.Value, 'reqprefix', _('Request IPv6-prefix'),
+						_('Either a prefix length hint (e.g. 56) only, whereby the operator selects the prefix, or specify a prefix also (e.g. <code>2001:db8::/56</code>)'));
 		o.value('auto', _('Automatic'));
		o.value('no', _('disabled'));
 		o.value('48');
 		o.value('52');
 		o.value('56');
```

systemcrash, 2024-11-30:
Closed by eadd59fc345a1d9fca66b200156bc382f603d12b

systemcrash, 2024-11-30:
Thanks for the fixes @ynezz and @try496

closed 2024-11-30 by systemcrash