openwrt/luci #7401: routes.js  The translation is not displaying

try496, 2024-11-17:
### How can we otherwise help?

I can see that Simplified Chinese and Polish are provided, but the translation is not displaying. This is after a fresh compilation and installation.

#: modules/luci-mod-network/htdocs/luci-static/resources/view/network/routes.js:96
#: modules/luci-mod-network/htdocs/luci-static/resources/view/network/routes.js:173
msgid ""
"A numeric table index, or symbol alias declared in %s. Special aliases local "
"(255), main (254) and default (253) are also valid"
msgstr "数值表索引，在 %s "
"中声明的符号别名。特殊别名local（255）、main（254）和default（253）是有效的"

#: modules/luci-mod-network/htdocs/luci-static/resources/view/network/routes.js:96
#: modules/luci-mod-network/htdocs/luci-static/resources/view/network/routes.js:173
msgid ""
"A numeric table index, or symbol alias declared in %s. Special aliases local "
"(255), main (254) and default (253) are also valid"
msgstr ""
"Indeks tablicy numerycznej lub alias symbolu zadeklarowany w %s. Specjalne "
"aliasy local (255), main (254) i default (253) są również prawidłowe"

[image]
[image]

try496, 2024-11-17:
Using OpenWrt 24.10 for the build.

hnyman, 2024-11-17:
It may be a coding error in 
https://github.com/openwrt/luci/blob/09e8d5108ed678db5b940b1eb51dff2b9b6af189/modules/luci-mod-network/htdocs/luci-static/resources/view/network/routes.js#L96

The translate function "_" has parentheses that cover also the contents of .format ...
The closing parentheses might need to be before .format

I am not sure if that is right, or should that just be split into three parts, two translated strings and untranslated code in between.

hnyman, 2024-11-17:
Yeah, looks like a recent error from @systemcrash in a9e1685

I will change the place of the erroneous parentheses:
`_('A numeric table index, or symbol alias declared in %s. Special aliases local (255), main (254) and default (253) are also valid'.format('<code>/etc/iproute2/rt_tables</code>'))`
-->
`_('A numeric table index, or symbol alias declared in %s. Special aliases local (255), main (254) and default (253) are also valid').format('<code>/etc/iproute2/rt_tables</code>')`

commit cfa774ee (2024-11-17): luci-mod-network: Fix for a9e1685 to enable translation

Fix the erroneous placement of ) in a9e1685 to actually provide translation.

Fixes #7401
closed 2024-11-17 by hnyman