openwrt/luci #7404: Web interface doesn't allow to set `option dhcpv4 'server'` when using odhcp

larskotthoff, 2024-11-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Actual behaviour

When setting up the DHCP server for an interface, there's no way to specify `option dhcpv4 'server'` when using odhcp. This results in the default value being used, which disables the DHCP server for the interface.

### Expected behaviour

UI allows to set the option.

### Steps to reproduce

1. Switch to odhcp.
2. Create a new interface and enable DHCP for it (default options ok).
3. Send a DHCP discover on that interface... and get nothing back.

### Additional Information

```shell
23.05.0
```

systemcrash, 2024-11-17:
You mean `-S` here?: https://thekelleys.org.uk/dnsmasq/docs/dnsmasq-man.html

That is covered in the init file. Although where would it go? Under DHCP or in the interface section?

larskotthoff, 2024-11-17:
No, this is when not using dnsmasq but odhcp instead. The missing option should be in the interface section.