openwrt/luci #7406: <package-name>: <summary>luci-app-p910nd

xh0823, 2024-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

[image]
[image]
[image]

### Actual behaviour

I have added kmod-lp
kmod-usb-printer
These two

### Expected behaviour

404 Not Found
Sorry, the object you requested was not found.

No page is registered at '/admin/system/opkg'. If this url belongs to an extension, make sure it is properly installed. If the extension was recently installed, try removing the /tmp/luci-indexcache file.

Unable to dispatch: /cgi-bin/luci/admin/system/opkg

### Steps to reproduce

Open error

### Additional Information

```shell
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28225-9b88505660"
OPENWRT_BOARD="mediatek/filogic"
OPENWRT_ARCH="aarch64_cortex-a53"
OPENWRT_TAINTS="no-all"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r28225-9b88505660"
OPENWRT_BUILD_DATE="1731665876"
```

dannil, 2024-11-18:
@systemcrash it directly invokes the old opkg package manager url `/admin/system/opkg` at https://github.com/openwrt/luci/blob/master/applications/luci-app-p910nd/htdocs/luci-static/resources/view/p910nd.js#L22 (among some other places) which doesn't work when the new `luci-app-package-manager` is located at [admin/system/package-manager](https://github.com/openwrt/luci/blob/master/applications/luci-app-package-manager/root/usr/share/luci/menu.d/luci-app-package-manager.json#L2)

systemcrash, 2024-11-19:
hi @xh0823 you can test my fix at:

https://github.com/systemcrash/luci/blob/p910nd_pkg/applications/luci-app-p910nd/htdocs/luci-static/resources/view/p910nd.js

xh0823, 2024-11-19:
> hi @xh0823 you can test my fix at:
> 
> https://github.com/systemcrash/luci/blob/p910nd_pkg/applications/luci-app-p910nd/htdocs/luci-static/resources/view/p910nd.js

I applied your code and it works fine now. Thank you.

closed 2024-11-19 by xh0823

systemcrash, 2024-11-19:
Fixed by b9496f2023cb6b5f51dff88f1cf3a3063390dae4